intel/compute-runtime #632: [ARC A380/A750/A770) Please fix these problem (OpenCL) [Important!]

MisterHM80, 2023-03-26:
It's about my OpenCL benchmark.

"Async Compute" test no longer works since driver version 4000. To 3975 went without problems.

With 4000 version, this only works if I reduce to 512, 512.

Equally buggy runs from the start with "HDR" and "TEXTURE"
If you run in order:
TEXTURE -> HDR -> TEXTURE -> HDR

Then we see the time indication for each 2nd run higher time.

With Nvidia or AMD there was no such behavior until now.

I would be very grateful if this could finally be fixed.
Even if priority is last with OpenCL.

Download (HardwareLuxx Forum)
https://www.hardwareluxx.de/community/threads/28-01-2023-r9-fury-added-amd-nvidia-intel-mrh-opencl-benchmark-rankliste.1302003/

Note : The same was already posted on Discord some time ago.

JablonskiMateusz, 2023-03-28:
Hi @MisterHM80 
could you provide driver version reported by clinfo? I don't know how to map `driver version 4000` to our release

MisterHM80, 2023-03-28:
Meant between 31.0.101.4032 and 31.0.101.4255 which doesn't work with Async Compute. 31.0.101.3975 or older works as it should.

Or have I misunderstood that?

MisterHM80, 2023-03-28:
No. I don't see any point in it either.
The 5 other tests are running. Just not Async Compute since 4032 version.

Had just installed again 3975 and it works again.
I don't need to try anything.
Please try it to fix it.

MisterHM80, 2023-05-11:
@JablonskiMateusz

Any news on this?

Wagnard, 2023-05-17:
Hi, this is now fixed with Intel driver 31.0.101.4369

MisterHM80, 2023-05-17:
Thank you for your feedback.
I'm glad that's fixed now with Async Compute :)

I hope, the HDR Bug will be fixed soon too
That make on 2nd Run slower (Bicubic, Texture and Async Compute), if before run first time HDR.
Same for HDR after 2nd Run.

MisterHM80, 2023-12-20:
@JablonskiMateusz
What about the HDR bug now? Still exists until now.
Start the Tool and 1.st run with HDR. 2.nd run is up to 33% slower.

MisterHM, 2025-01-04:
B580 dont have HDR Bug like A770.
But i am not happy about Async Compute Performance on both.
Why so slow?

MisterHM80, 2025-01-04:
B580 dont have HDR Bug like A770.
But i am not happy about Async Compute Performance on both.
Why so slow?

MisterHM80, 2026-08-17:
Any ETA when this fix will reach public Arc B‑Series drivers?

jwarchul, 2026-08-18:
Hi @MisterHM80,

We sincerely apologize for the delayed response.

* **HDR Bug:** The fix for this issue is already available in the [latest driver version - `32.0.101.8974`](https://www.intel.com/content/www/us/en/download/785597/intel-arc-graphics-windows.html). Could you please confirm it?

* **Async Compute Performance:** The issue has been identified and the fix has already been merged. It should be included in the next driver release. We will update you here as soon as it becomes available.

* **GridView Issue:** Could you please provide more details regarding this problem, along with the steps to reproduce it?

Thank you for your valuable contribution, and apologies once again for the long wait.

MisterHM80, 2026-08-18:
@jwarchul 

HDR:  
On the iGPU (A‑Series), the previous slowdown after the second run no longer occurs once the tool has been opened. The HDR behavior is now stable on the A‑Series.

GridView:  
I created a video demonstrating the different behavior between the iGPU (A‑Series) and the ARC PRO B70 (B‑Series). You can clearly see that focus switching between my MrH OpenCL Benchmark and the Windows timer works smoothly on the A‑Series, while the B‑Series shows visible slowdowns: when the benchmark window has focus, the Windows timer increments much more slowly and the system responds sluggishly; when the timer window has focus, the B‑Series behaves normally.
https://www.youtube.com/watch?v=QlFW5t4FI4I

MisterHM80, 2026-08-19:
@jwarchul 

Was the significant Texture performance improvement in driver 32.0.101.8974 (ARC A770) intentional?
(8864 → 8974: Texture Extreme 103 ms → 42 ms)

jwarchul, 2026-08-20:
Hi @MisterHM80,

Thank you for the additional information. We will continue investigating this issue internally and will update you as soon as we have any findings.

Thanks for your cooperation.

MisterHM80, 2026-10-03:
@jwarchul 

> * **Async Compute Performance:** The issue has been identified and the fix has already been merged. It should be included in the next driver release. We will update you here as soon as it becomes available.

Any update on this? Up to the current 9034 driver I still see no improvement in async compute on A770 and B70 (both installed together, iGPU disabled).

MisterHM80, 2026-10-13:
> [@jwarchul](https://github.com/jwarchul)
>
>
> Any update on this? Up to the current 9034 driver I still see no improvement in async compute on A770 and B70 (both installed together, iGPU disabled).

The latest driver (9150) successfully resolves the async compute issue, confirming a significant performance improvement on both my A770 and B70. :)

- Arc A770: 1557 > 4492 (Extreme) / 376 > 1127 (Godmode)
- Arc PRO B70: 872,2 > 3382 (Extreme) / 219,2 > 846 (Godmode)