intel/compute-runtime #633: Unable to install compute-runtime

kurapov-peter, 2023-04-11:
Installing the latest release results in following:
```
(Reading database ... 4504 files and directories currently installed.)
Preparing to unpack intel-igc-core_1.0.13230.7_amd64.deb ...
Unpacking intel-igc-core (1.0.13230.7) over (1.0.12812.24) ...
Preparing to unpack intel-igc-opencl_1.0.13230.7_amd64.deb ...
Unpacking intel-igc-opencl (1.0.13230.7) over (1.0.12812.24) ...
Preparing to unpack intel-level-zero-gpu_1.3.25593.11_amd64.deb ...
Unpacking intel-level-zero-gpu (1.3.25593.11) over (1.3.25018.24) ...
Preparing to unpack intel-opencl-icd_23.05.25593.11_amd64.deb ...
Unpacking intel-opencl-icd (23.05.25593.11) over (22.49.25018.24) ...
Preparing to unpack libigdgmm12_22.3.0_amd64.deb ...
Unpacking libigdgmm12:amd64 (22.3.0) over (22.3.0) ...
Setting up intel-igc-core (1.0.13230.7) ...
Setting up intel-igc-opencl (1.0.13230.7) ...
dpkg: dependency problems prevent configuration of intel-opencl-icd:
 intel-opencl-icd depends on ocl-icd-libopencl1; however:
  Package ocl-icd-libopencl1 is not installed.

dpkg: error processing package intel-opencl-icd (--install):
 dependency problems - leaving unconfigured
Setting up libigdgmm12:amd64 (22.3.0) ...
Setting up intel-level-zero-gpu (1.3.25593.11) ...
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
Errors were encountered while processing:
 intel-opencl-icd
```

Reproduction steps: https://github.com/intel/compute-runtime/issues/595#issuecomment-1497677922

JablonskiMateusz, 2023-04-11:
Dependency on `ocl-icd-libopencl1` was added in https://github.com/intel/compute-runtime/commit/543c854a3b7fc2590a0e0b72ad3260066ecfea89 (2 years ago) as neo runtime is useless without loader.
More details: https://github.com/intel/compute-runtime/issues/470

kurapov-peter, 2023-04-11:
@JablonskiMateusz, could you elaborate, please? Does it mean that the installation instruction is not complete and the dependency is required? Is there a recommended installation way? (I'm not using opencl, but level zero directly).

JablonskiMateusz, 2023-04-12:
our packages adds support for two APIs - OpenCL and L0.
There are dependencies for both of them - gmmlib and igc which are included in the instructions.

For OpenCL, our runtime package (`intel-opencl-icd`) requires `ocl-icd-libopencl1` as OpenCL ICD loader is the only way to use our runtime.

For Level Zero, our runtime package is `intel-level-zero-gpu`.

Additionally, as a part of OpenCL package we provide `ocloc` tool which may be useful in Level Zero cases to compile kernels from source.

Assuming you are not using OpenCL API, my recommendation depends on your need for ocloc tool. 
If you need ocloc for kernels compilation, you need to install `ocl-icd-libopencl1` as a dependency for `intel-opencl-icd`.
If you don't need ocloc, you just don't need to install `intel-opencl-icd`.

kurapov-peter, 2023-04-12:
@JablonskiMateusz, thanks for the clarifications! Ocloc is a nice-to-have tool for quick debugging but is not required.

We install all the packages from a release as described in the release notes. Here's what it looks like: https://github.com/intel-ai/hdk/blob/main/docker/Dockerfile.l0. Seems like we were just lucky not to hit the problem because some of the dependencies we install here https://github.com/intel-ai/hdk/blob/main/docker/Dockerfile are also installing the `ocl-icd-libopencl1`, masking the problem. The very same installation steps on clean Ubuntu fail.

JablonskiMateusz, 2023-04-21:
@kurapov-peter  can we close this issue?

closed 2023-04-28 by JablonskiMateusz