intel/compute-runtime #636: A770: Possible miscompilation of mapreduce-like code

maleadt, 2023-04-20:
Working on Julia support for oneAPI, I've isolated a test failure that only occurs on my A770 to the following Julia code:

```julia
using oneAPI

# complete reduction values by a group, using local memory for communication
@inline function partialsum_group(val::T, neutral) where {T}
    items = get_local_size(0)
    item = get_local_id(0)

    # local mem for a complete reduction
    shared = oneLocalArray(T, (1024,))
    @inbounds shared[item] = val

    # perform a reduction
    d = 1
    while d < items
        barrier()
        index = 2 * d * (item-1) + 1
        @inbounds if index <= items
            other_val = if index + d <= items
                shared[index+d]
            else
                neutral
            end
            shared[index] = shared[index] + other_val
        end
        d *= 2
    end

    # load the final value on the first item
    if item == 1
        val = @inbounds shared[item]
    end

    return val
end

# partial reduction of the input vector, using a grid-stride loop
function partialsum(elements, reduced, input)
    localIdx = get_local_id(0)
    localDim = get_local_size(0)
    groupIdx = get_group_id(0)
    groupDim = get_num_groups(0)

    @inbounds begin
        # load the neutral value
        #
        # for this MWE, the value is always 0, but hard-coding it makes the bug occur
        # less often.
        neutral = reduced[groupIdx]
        val = neutral + neutral

        # reduce serially across chunks of the input vector.
        #
        # for this MWE, we only execute the loop body once (as we allocate exactly
        # items * groups elements), but removing the loop makes the bug happen less often.
        ireduce = localIdx + (groupIdx - 1) * localDim
        while ireduce <= elements
            val = val + input[ireduce]
            ireduce += localDim * groupDim
        end

        # reduce all values within the group
        val = partialsum_group(val, neutral)

        # write back to memory
        if localIdx == 1
            reduced[groupIdx] = val
        end
[... 576 lines cut ...]
        fprintf(stderr, "Error allocating memory for SPIR-V binary\n");
        exit(EXIT_FAILURE);
    }

    if (fread(*spirv, 1, *spirv_len, file) != *spirv_len) {
        fprintf(stderr, "Error reading SPIR-V binary\n");
        exit(EXIT_FAILURE);
    }

    fclose(file);
}
```

Running this code produces similar results:

```
❯ gcc -g -o partialsum partialsum.c -lze_loader && ./partialsum partialsum.spv
Int32[800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 784, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 784, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, 800, ]
```

Also note that I'm doing some seemingly redundant sequences of closing/executing the command list and synchronizing the command queue in between memory initialization, kernel launch, and memory readback; I'm not sure why those are needed, as I expected operations on command lists to be ordered? I'd appreciate some clarification here 🙂

All of the above has been tested on Linux, kernel 6.2, with an A770 and using both the drivers packages by Arch Linux (compute-runtime 22.43.24595.30 and graphics compiler 1.0.12812.26) and our own builds (compute-runtime 22.53.25593 and graphics compiler 1.0.13230).

[partialsum.spv.zip](https://github.com/intel/compute-runtime/files/11286683/partialsum.spv.zip)

HoppeMateusz, 2023-04-21:
> Also note that I'm doing some seemingly redundant sequences of closing/executing the command list and synchronizing the command queue in between memory initialization, kernel launch, and memory [...]

Command lists do not guarantee in-order completion of appended commands, only start of command is in-order unless RELAXED ordering is requested

https://spec.oneapi.io/level-zero/latest/core/PROG.html#appending

"By default, commands are started in the same order in which they are appended. However, an application may allow the driver to optimize the ordering by using [ZE_COMMAND_LIST_FLAG_RELAXED_ORDERING](https://spec.oneapi.io/level-zero/latest/core/api.html#ze-command-list-flags-t). Reordering is guaranteed to only occur between barriers and synchronization primitives."

to ensure commands are executed in order, BARRIER is needed:

"Execution Barriers - used to communicate execution dependencies between commands within a command list or across command queues, devices and/or Host."

```
// Append a barrier into a command list to ensure hKernel1 completes before hKernel2 begins
zeCommandListAppendBarrier(hCommandList, nullptr, 0, nullptr);
```

maleadt, 2023-04-21:
Ah interesting, guess I was used to the CUDA stream model. Thanks for clarifying!

HoppeMateusz, 2023-04-21:
you are correct
 
replacing:

```
        // Execute and wait for pending commands
        // XXX: why is this needed?
        result = zeCommandListClose(cmd_list);
        assert(result == ZE_RESULT_SUCCESS);
        result = zeCommandQueueExecuteCommandLists(queue, 1, &cmd_list, NULL);
        assert(result == ZE_RESULT_SUCCESS);
        result = zeCommandQueueSynchronize(queue, UINT32_MAX);
        assert(result == ZE_RESULT_SUCCESS);
        result = zeCommandListReset(cmd_list);
        assert(result == ZE_RESULT_SUCCESS);

```
with:

`        zeCommandListAppendBarrier(cmd_list, nullptr, 0, nullptr);`

should logically be the same.

we need to investigate this issue.

zzdanowicz, 2023-04-21:
@maleadt have you tried to replace UINT32_MAX with UINT64_MAX as timeout value?

maleadt, 2023-04-21:
Yes, that doesn't help.

zzdanowicz, 2023-04-21:
@maleadt I know this might be also something in the driver, but in the function: `partialsum_group` don't you think `barrier()` call is needed before

```
if item == 1
        val = @inbounds shared[item]
end
```
As the barrier call in while loop will not be executed after the last iteration.

maleadt, 2023-04-21:
> As the barrier call in while loop will not be executed after the last iteration.

The last iteration should only ever be executed by the first thread in a group (i.e. `item == 1`), which is then also the thread that goes and read the value produced by that iteration from shared memory, so I don't think an additional barrier is needed. In any case, I tested it, and it doesn't make a difference here.

zzdanowicz, 2023-04-25:
If you are willing to build a driver to quickly test a possible fix, then let me know, you'll need to add two lines of code.

maleadt, 2023-04-25:
Sure, I can do that.

zzdanowicz, 2023-04-25:
Please find the file: level_zero/core/source/cmdqueue/cmdqueue_hw.inl
method: CommandQueueHw<gfxCoreFamily>::dispatchTaskCountPostSyncRegular
and add these two lines:
```
    args.hdcPipelineFlush = true;
    args.unTypedDataPortCacheFlush = true
```

maleadt, 2023-04-25:
So:

```patch
diff --git a/level_zero/core/source/cmdqueue/cmdqueue_hw.inl b/level_zero/core/source/cmdqueue/cmdqueue_hw.inl
index 5acb3e5633..6911fc653a 100644
--- a/level_zero/core/source/cmdqueue/cmdqueue_hw.inl
+++ b/level_zero/core/source/cmdqueue/cmdqueue_hw.inl
@@ -1115,6 +1115,8 @@ void CommandQueueHw<gfxCoreFamily>::dispatchTaskCountPostSyncRegular(
     args.dcFlushEnable = this->csr->getDcFlushSupport();
     args.workloadPartitionOffset = this->partitionCount > 1;
     args.notifyEnable = this->csr->isUsedNotifyEnableForPostSync();
+    args.hdcPipelineFlush = true;
+    args.unTypedDataPortCacheFlush = true;
     NEO::MemorySynchronizationCommands<GfxFamily>::addBarrierWithPostSyncOperation(
         cmdStream,
         NEO::PostSyncMode::ImmediateData,
```

That didn't help (I ensured the modified `libze_intel_gpu` is being loaded by checking with GDB).