intel/compute-runtime #641: Abort from device_binary_format/patchtokens_decoder

abagusetty, 2023-04-26:
I am running a HPC workload with PVC in explicit-scaling mode using SYCL API (via L0 plugin) and was wondering what could possibly have triggered the following error. 

It was bit hard to debug the error from the application side since the application was quite complex, and the error prompt didn't quite reveal anything useful. Is this some how related to excessive allocation of device-memory ? Just wanted to reach out for pointers before I dive into manual debugging. 

Version: 23.09.25812.14
Device: PVC (explicit-scaling mode)

```bash
Abort was called at 49 line in file:
/..../intel-gpu-umd/driver/intel-compute-runtime/shared/source/device_binary_format/patchtokens_decoder.cpp
```
Corresponds to: https://github.com/intel/compute-runtime/blob/master/shared/source/device_binary_format/patchtokens_decoder.cpp#L49

JablonskiMateusz, 2023-04-28:
Hi @abagusetty  could you attach kernel binary or repro steps? It looks like invalid binary

dimitryn, 2024-04-08:
Hi @JablonskiMateusz  , I getting same error on our Ubuntu 2024 machine. 
I using openvino 2024 and trying to run our model on GPU.
[image]
We using Neo release version [24.05.28454.6](https://github.com/intel/compute-runtime/releases/tag/24.05.28454.6)
Please advise how to resolve ?