intel/compute-runtime #643: A critical bug happened with HDR(?) Mapping with opencl

ShCby, 2023-05-11:
I am using proxmox ve host and using Ubuntu 22.04 LTS VM.

UHD770 enables SR-IOV.

when they use the decoding function using opencl in ubuntu, it have these video errors.

[image]

nyanmisaka, 2023-05-11:
It's a Plex exclusive issue and should be reported here.
See also https://forums.plex.tv/t/movie-corruption-when-plex-decodes-4k-hdr/834529

ShCby, 2023-05-11:
> It's a Plex exclusive issue and should not be reported here. See also https://forums.plex.tv/t/movie-corruption-when-plex-decodes-4k-hdr/834529

You can see more information here.
https://github.com/strongtz/i915-sriov-dkms/issues/57

nyanmisaka, 2023-05-12:
@ShCby As far as I know, intel hasn't officially supported SR-IOV in mainline kernels. So It makes sense that they won't test [media-driver](https://github.com/intel/media-driver) and [compute-runtime](https://github.com/intel/compute-runtime) with third-party modified DKMS in such use case.

The broken image seems to be caused by a mismatched DRM format modifier that used to describe the memory layout.
You can turn to https://github.com/intel/media-driver for help.

closed 2023-05-15 by ShCby