intel/compute-runtime #644: Incorrect installation instructions for 23.13.26032.30

al42and, 2023-05-17:
[There are no DEB packages for IGC 1.0.13700.14](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.13700.14)

```console
$ wget https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.13700.14/intel-igc-core_1.0.13700.14_amd64.deb
--2023-05-17 14:53:50--  https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.13700.14/intel-igc-core_1.0.13700.14_amd64.deb
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-05-17 14:53:50 ERROR 404: Not Found.
```

pszymich, 2023-05-17:
Hi, IGC has posted the release: https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.13700.14

al42and, 2023-05-17:
Thanks, but now sha256 signatures are out-of-date

closed 2023-05-22 by psidor99