intel/compute-runtime #648: Redefinition of APIENTRY macro

tpkessler, 2023-06-02:
Hi! The macro `APIENTRY` is defined both in `third_party/DirectX-Headers/include/wsl/winadapter.h` and `third_party/opengl_headers/GL/gl.h`. This causes a compiler warning with gcc12 and, owing to `-Werror`, a build error.

JablonskiMateusz, 2023-06-05:
Hi @tpkessler 
Could you share build log when it fails?

tpkessler, 2023-06-12:
```
[ 11%] Building CXX object level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/module/module_imp_helper.cpp.o
In file included from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/shared/source/os_interface/windows/windows_wrapper.h:31,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/shared/source/os_interface/windows/d3dkmthk_wrapper.h:10,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/shared/source/gmm_helper/gmm_lib.h:13,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/extensions/public/cl_gl_private_intel.h:11,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/source/sharings/gl/linux/gl_sharing_linux.h:10,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/source/sharings/gl/linux/gl_context_guard_linux.cpp:9:
/build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/third_party/DirectX-Headers/include/wsl/winadapter.h:78: error: "APIENTRY" redefined [-Werror]
   78 | #define APIENTRY
      |
In file included from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/source/sharings/gl/gl_sharing.h:15,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/source/sharings/gl/gl_context_guard.h:10,
                 from /build/intel-compute-runtime/src/compute-runtime-23.13.26032.30/opencl/source/sharings/gl/linux/gl_context_guard_linux.cpp:8:
/usr/include/GL/gl.h:83: note: this is the location of the previous definition
   83 | #define APIENTRY GLAPIENTRY
      |

```

commit e1b429db (2023-06-16): fix: redefinition of APIENTRY macro

Add guard for redifinition of APIENTRY in DirectX headers that could be
defined in other headers of this project like GL.

Fixes #648