intel/compute-runtime #651: compute-runtime-23.13.26032.30 core dumped

kreaninw, 2023-06-13:
When I run `clinfo`, it returns core dumped:

```
Abort was called at 36 line in file:
/home/abuild/rpmbuild/BUILD/compute-runtime-23.13.26032.30/shared/source/built_ins/built_ins.cpp
Aborted (core dumped)
```

And I can't enable OpenCL in LibreOffice (it worked before), Darktable crashes on opening (with OpenCL enabled).

CPU: Intel Core i5-8250U
GPU: Intel UHD Graphics 620 (KBL GT2), Mesa 23.1.2-350.1
OS: openSUSE Tumbleweed, kernel 6.3.6-1-default, Wayland
`intel-opencl`: 23.13.26032.30-1.1
`libigc1`: 1.0.13700.14-1.2
`libSPIRV-Tools`: 2023.3~rc1-1.1

I also reported this bug on **[Bugzilla SUSE Bug 1212193](https://bugzilla.suse.com/show_bug.cgi?id=1212193)**.

JablonskiMateusz, 2023-06-14:
Hi @kreaninw what version of igdfcl have you installed?

kreaninw, 2023-06-15:
Hi @JablonskiMateusz My `igdfcl` version is 1.0.13700.14-1.2.

JablonskiMateusz, 2023-06-15:
Please attach strace log and double check if dependencies for compiler libraries are satisfied (by running `ldd` command)

kreaninw, 2023-06-17:
The latest snapshot on openSUSE Tumbleweed fixed this issue.

closed 2023-06-17 by kreaninw