intel/compute-runtime #652: Newer 23.13 runtime causes: OpenCL error happened: clBuildProgram, Code: -6

jongames, 2023-06-15:
With the latest runtime 23.13 from the Intel repo installed on Ubuntu 23.04 Lunar Lobster, kernel version 6.2.6, running clinfo completes just normally without any indication of errors for the driver/runtime, but trying to run any OpenCl accelerated workload such as tonemapping through ffmpeg or [clgpustress](https://github.com/matszpk/clgpustress) results in some form of this error: 
`OpenCL error happened: clBuildProgram, Code: -6` where code -6 seems to be an out of memory error from my search online.  

Reverting to the previous intel opencl runtime 23.05, fixes all issues and both previously mentioned programs run without errors: 
`apt-cache policy intel-opencl-icd
intel-opencl-icd:
  Installed: 23.05.25593.18-601~22.04
  Candidate: 23.13.26032.26-627~22.04
  Version table:
     23.13.26032.26-627~22.04 500
        500 https://repositories.intel.com/graphics/ubuntu jammy/arc amd64 Packages
 *** 23.05.25593.18-601~22.04 500
        500 https://repositories.intel.com/graphics/ubuntu jammy/arc amd64 Packages
        100 /var/lib/dpkg/status
`

My system is using the i7-12700K CPU's integrated Intel UHD 770 graphics for opencl acceleration with plenty of free RAM (64GB installed and only 25GB used) so it is not really a memory issue with my system but the newer runtime acts differently somehow.

JablonskiMateusz, 2023-06-19:
Hi @jongames 
could you try more recent intel-opencl-icd  with older igc packages?

jongames, 2023-06-23:
> Hi @jongames could you try more recent intel-opencl-icd with older igc packages?

Apparently, my system was using an older IGC library that I had manually installed from github before attempting to use the Intel APT repo which I had found later on: 

```
intel-igc-opencl:
  Installed: 1.0.13463.18
  Candidate: 1.0.13463.18
  Version table:
 *** 1.0.13463.18 100
        100 /var/lib/dpkg/status
intel-igc-core:
  Installed: 1.0.13463.18
  Candidate: 1.0.13463.18
  Version table:
 *** 1.0.13463.18 100
        100 /var/lib/dpkg/status
```

Now I decided to just remove those above mentioned packages and everything works...

The only other "igc" named package on my system that is actually from the Intel APT repo is this: 
`intel-igc-cm is already the newest version (1.0.176+i600~22.04)`
Which I guess is what I should have been using, instead of those other problematic packages.  

Thank you for suggesting it could be an issue with "igc" as I didn't realize those manually installed packages were still there.

closed 2023-06-23 by jongames