intel/compute-runtime #656: Device name reported with clinfo is not Processor Graphics Name per ark.

Toby-Broom, 2023-06-21:
Can you update the libraries so that your resolve the hex code to the correct Processor Graphics name e.g.

Device Name                                     Intel(R) Graphics [0xa780]
Device Name                                     Intel® UHD Graphics 770

JablonskiMateusz, 2023-06-22:
Hi @Toby-Broom  could you share you driver version?

Toby-Broom, 2023-06-22:
Version: 23.0.2

JablonskiMateusz, 2023-06-22:
please share output from  `clinfo | grep -i version`

Toby-Broom, 2023-06-22:
Platform Version                                OpenCL 3.0
  Platform Extensions with Version                cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
  Platform Numeric Version                        0xc00000 (3.0.0)
  Device Version                                  OpenCL 3.0 NEO
  Device Numeric Version                          0xc00000 (3.0.0)
  Driver Version                                  22.43.24595
  Device OpenCL C Version                         OpenCL C 1.2
  Device OpenCL C all versions                    OpenCL C                                                         0x400000 (1.0.0)
    IL version                                    SPIR-V_1.2
    ILs with version                              SPIR-V                                                           0x402000 (1.2.0)
    SPIR versions                                 1.2
  Built-in kernels with version                   (n/a)
  Device Extensions with Version                  cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
  ICD loader Version                              2.3.1

JablonskiMateusz, 2023-06-22:
Seems you have too old driver. The change was merged at https://github.com/intel/compute-runtime/commit/375eb35697eb982743816f812bf8391ce1ae3ef8 which is included in 22.49 release, while you have 22.43 release.

Toby-Broom, 2023-06-22:
OK great thanks for the feeback

closed 2023-06-22 by Toby-Broom