intel/compute-runtime #659: cl_khr_gl_sharing GLX support

smunaut, 2023-07-04:
As discussed in #166  , although the extension has been added recently, only EGL context are supported and having GLX support is required for some applications.

pearsche, 2023-07-05:
One of the applications being Davinci Resolve

k1gen, 2023-07-25:
I'm interested in this too, because currently there's no way to use Davinci Resolve on Linux, even though EGL support (which took 4 years to develop) is merged. Waiting for any information on this!

AdamCetnerowski, 2023-07-31:
We are open to include this feature in the driver. We are not able to devote engineering effort to implementation at this time, but would welcome contributions from the community.

I'd also like to note that we recently added support for [cl_khr_external_memory](https://registry.khronos.org/OpenCL/specs/3.0-unified/html/OpenCL_Ext.html#cl_khr_external_memory), which should show up in the August release.

smunaut, 2023-08-10:
FWIW in the branch I pushed in #667  , I have now added GLX support as well.

smunaut, 2023-08-10:
So the good new is that Resolve starts and you see something in the viewers now : https://i.imgur.com/qyfOARh.png
The bad news is that it only does that when running in valgrind ... which like the name suggests, brings it to a grind :sweat:

smunaut, 2023-08-10:
Found a workaround ... I had to remove the RTLD_DEEPBIND option used in intel-compute to load OsLibraries and then it starts.
I'm not entirely sure what the issue is or what the consequences are of removing this, so still some check needed.

But now I can start resolve and see stuff in the viewers. It's a bit glitchy and I get some weird errors :

```
0x7f72886e4680 | IO                   | ERROR | 2023-08-10 23:23:50,585 | Failed to decode clip </tmp/osmodevcall-20210326-fixeria-osmoran-status-update_vp9.webm>, track 0, frame 27687: Requested position is outside of track bound.
0x7f72896e6680 | IO                   | ERROR | 2023-08-10 23:23:50,595 | Failed to decode clip </tmp/osmodevcall-20210326-fixeria-osmoran-status-update_vp9.webm>, track 0, frame 27805: Requested position is outside of track bound.
0x7f72886e4680 | IO                   | ERROR | 2023-08-10 23:23:50,618 | Failed to decode clip </tmp/osmodevcall-20210326-fixeria-osmoran-status-update_vp9.webm>, track 0, frame 27914: Failed to read the video frame from disk.
0x7f72886e4680 | IO                   | ERROR | 2023-08-10 23:23:50,655 | Failed to decode clip </tmp/osmodevcall-20210326-fixeria-osmoran-status-update_vp9.webm>, track 0, frame 28032: Failed to read the video frame from disk.
0x7f7288ee5680 | IO                   | ERROR | 2023-08-10 23:23:53,658 | Failed to decode clip </tmp/osmodevcall-20210326-fixeria-osmoran-status-update_vp9.webm>, track 0, frame 28149: Failed to read the video frame from disk.
```

But heh, progress !

SimonBrandner, 2023-08-10:
(just wanted to say thank you! - awesome to see someone trying to get Resolve working!)

k1gen, 2023-08-11:
@smunaut hey, I'm on Arch and have a UHD620. what should I do to test?

SimonBrandner, 2023-08-11:
/me is on Nix; might be interesting trying to figure out how to test this :D

smunaut, 2023-08-11:
So you need to build :
* https://gitlab.freedesktop.org/246tnt/mesa/-/tree/clgl-export?ref_type=heads
* https://github.com/smunaut/compute-runtime/tree/clgl

(make sure to checkout the correct branch as pointed in those links)

You can install in a temporary prefix like /tmp/test_root or something.
And then :

``
export OCL_ICD_VENDORS=/tmp/_root/etc/OpenCL/vendors/
export LD_LIBRARY_PATH=/tmp/_root/lib64
export NEOReadDebugKeys=1
export DisableDeepBind=1 
```

* Make sure the .icd gets installed in that subdirectory, IIRC I had to do something more than set the prefix for that but don't remember. You can also create that file by hand, it's just a text file pointing to the right .so
* Beware of `lib` vs `lib64` depends on your distro I think.

Then just try to start resolve.

smunaut, 2023-08-11:
btw, I'm on the unofficial blackmagic discord and on the freedesktop one, so you can ping me there username `246tnt` which might be easier than github issues. Also on IRC as `tnt` on libera.

smunaut, 2023-08-24:
Ah will do. But gimme some time, I haven't actually pushed the latest stuff :sweat_smile:  
I'll do that tonight.

smunaut, 2023-08-24:
Ok pushed.

You need the `23.2-resolve` branch from my MESA tree : https://gitlab.freedesktop.org/246tnt/mesa/-/tree/23.2-resolve?ref_type=heads
And the `clgl` branch of this tree (`compute-runtime`).

specialworld83, 2023-10-10:
I tested it on Fedora 38 and I can say that it works perfectly.

smunaut, 2023-10-10:
@specialworld83 Thanks for testing !

smunaut, 2023-10-12:
Do you use IRC or discord ? Would be easier to do this in a "live" session :)

smunaut, 2023-10-12:
We can probably create a thread in #resolve on the BMD discord : https://discord.com/invite/blackmagic-design-community-479297254528647188

smunaut, 2023-10-13:
So just to update the issue here. There were 2 bugs identified :
 - One in Mesa itself where it export the texture reporting it as being linear while it is in fact tiled ... The export code doesn't handle GFX12.5+ (Alchemist) cards properly since they have a new tiling mode ...
 - One in the new code where the tiling mode is not supported and I'm not sure how to set it.

Note however that theses are not regression, it doesn't work any better in the current cl_khr_gl_sharing support that's currently merged in, so theses patches are still an improvement.

smunaut, 2023-10-13:
@JablonskiMateusz   @eero-t Maybe one of you knows : How can I specify which tiling mode to use ?  I know which one is used by mesa, but now I need to tell the compute stack about it ...  I found a `linearStorage` flag but that's a bit restrictive, I need to specify some specific mode ...   (and then there is a aux stuff but ... one thing at a time)

Leocraft1, 2023-11-13:
Any update? How do I install the patch for OpenCL libraries?

smunaut, 2023-11-13:
There is an open PR I need to address a couple comments on it and update it to latest code, but I'm a bit busy with life stuff ATM so might take a few weeks until I get to it.

eero-t, 2023-11-21:
> @JablonskiMateusz @eero-t Maybe one of you knows : How can I specify which tiling mode to use ? I know which one is used by mesa, but now I need to tell the compute stack about it ... I found a [...]

Sorry, unfortunately I don't know either (I'm not driver developer / related to this project, just an occasional bug commenter).

closed 2024-03-01 by Compute-Runtime-Automation

smunaut, 2024-03-01:
@SimonBrandner  Hopefully that means #673 will be merged imminently :sweat_smile:

SimonBrandner, 2024-03-02:
Awesome!

RannyBergamotte, 2024-03-23:
Thanks so much smunaut and kabdalla for y'alls amazing work, I've been eyeing this issue for 2 years now, so glad to see it fixed, can't wait for to arrive in the arch repos in a few months from now! Truly thank you!

wyup, 2024-03-27:
Hi, I'm running Fedora 41 rawhide with i5-13500 iGPU, following [this guide](https://www.codelinsoft.it/sito/blog/installare-da-vinci-resolve-con-scheda-video-iris-xe-graphics-su-fedora-38.html) for building mesa branch 23.3-resolve, I get compilation error:

ninja -C build64 install

```
../src/amd/llvm/ac_llvm_helper.cpp: In function ‘LLVMOpaquePassManager* ac_create_passmgr(LLVMTargetLibraryInfoRef, bool)’:
../src/amd/llvm/ac_llvm_helper.cpp:311:28: error: ‘createLoopSinkPass’ no se declaró en este ámbito
311 | unwrap(passmgr)->add(createLoopSinkPass());
| ^~~~~~~~~~~~~~~~~~
[112/949] Compiling C object src/amd/vulkan/libvulkan_radeon.so.p/radv_cmd_buffer.c.o
ninja: build stopped: subcommand failed.
```

smunaut, 2024-03-27:
@wyup Don't bother with that branch anymore, just build latest mesa release, it has everything you need in it.

wyup, 2024-03-28:
Any chance to add compatibility for intel integrated GPU detection for Resolve by installing corresponding binary rpm packages?
I already have mesa, compute-runtime and cl-gc:

I mean, I have latest mesa, intel-media-driver, intel-compute-runtime, intel-opencl, intel-igc, oneVPL-intel-gpu, mesa-libGL, mesa-libOpenCL.... what else do i need not to build?

smunaut, 2024-03-28:
Which distribution ?

The latests release of intel-compute-runtime doesn't have the patch yet. It's been merged in master but there was no stable release since. The only distribution to have release a package including the patch is fedora and this happened like ... 12h ago or so.

wyup, 2024-03-28:
I'm on fedora rawhide 41

smunaut, 2024-03-28:
@wyup See https://src.fedoraproject.org/rpms/intel-compute-runtime  there is now [intel-compute-runtime-24.09.28717.12-1.fc41](https://koji.fedoraproject.org/koji/search?terms=intel-compute-runtime-24.09.28717.12-1.fc41&type=build&match=exact)  since this morning.

wyup, 2024-03-28:
I've upgraded intel-compute-runtime, intel-opencl, intel-level-zero, intel-ocloc and intel-opencl to 24.09.28717.12-1.fc41 but Resolve does not detect iGPU.

smunaut, 2024-03-28:
What does `clinfo` say ?

wyup, 2024-03-28:
```
Number of platforms                               2
  Platform Name                                   rusticl
  Platform Vendor                                 Mesa/X.org
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_create_command_queue cl_khr_expect_assume cl_khr_extended_versioning cl_khr_icd cl_khr_il_program cl_khr_spirv_no_integer_wrap_decoration
  Platform Extensions with Version                cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_create_command_queue                                      0x400000 (1.0.0)
                                                  cl_khr_expect_assume                                             0x400000 (1.0.0)
                                                  cl_khr_extended_versioning                                       0x400000 (1.0.0)
                                                  cl_khr_icd                                                       0x400000 (1.0.0)
                                                  cl_khr_il_program                                                0x400000 (1.0.0)
                                                  cl_khr_spirv_no_integer_wrap_decoration                          0x400000 (1.0.0)
  Platform Numeric Version                        0xc00000 (3.0.0)
  Platform Extensions function suffix             MESA
  Platform Host timer resolution                  1ns

  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 24.0.3
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   rusticl
Number of devices                                 0

  Platform Name                                   Clover
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  rusticl
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   <error: 0 devices, multiple matching platforms!>
  clCreateContext(NULL, ...) [default]            No devices found in platform
  clCreateContext(NULL, ...) [other]              <error: no devices in non-default plaforms>
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.3.2
  ICD loader Profile                              OpenCL 3.0
```

smunaut, 2024-03-28:
What CPU do you have ?

wyup, 2024-03-28:
i5-13500

smunaut, 2024-03-28:
Maybe you're hitting : https://github.com/intel/compute-runtime/issues/710 ?  Are you on kernel 6.8.x ?

wyup, 2024-03-28:
I'm on 6.9.0-0.rc1.17.fc41.x86_64
it looks somebody made it with [this guide](https://www.codelinsoft.it/sito/blog/installare-da-vinci-resolve-con-scheda-video-iris-xe-graphics-su-fedora-38.html) on a intel iris Xe discrete GPU

eero-t, 2024-03-28:
> I'm on 6.9.0-0.rc1.17.fc41.x86_64 it looks somebody made it with [this guide](https://www.codelinsoft.it/sito/blog/installare-da-vinci-resolve-con-scheda-video-iris-xe-graphics-su-fedora-38.html) on a intel iris Xe discrete GPU

Then you are impacted by the bug.  Workaround:
```
export NEOReadDebugKeys=1
export OverrideGpuAddressSpace=48
```

smunaut, 2024-03-28:
And also add  `export DisableDeepBind=1`  or resolve will likely crash.

eero-t, 2024-03-28:
> And also add `export DisableDeepBind=1` or resolve will likely crash.

Could you mention that also on #710 ?

smunaut, 2024-03-28:
@eero-t That has nothing to do with #710  :sweat_smile:   You always need to do that with resolve and Intel ...

wyup, 2024-03-28:
it works!! you're a hero :-)

vidalinux, 2024-04-12:
Using this patch on Arch linux + Davinci Resolve with latest intel-compute-runtime, everything works good but video looks like this; 
[image]

smunaut, 2024-04-12:
@vidalinux I assume you're on intel Arc ?
If so, yeah there is a bug ... but I don't have access to Arc hardware so I can't debug/fix it.

vidalinux, 2024-04-12:
@smunaut ok thanks for your answer. maybe @intel can donate one ;)

wyup, 2024-04-19:
Resolve doesn't start anymore after Fedora update:
```
diegohum@tower:~$ resolve
ActCCMessage Already in Table: Code= c005, Mode= 13, Level=  1, CmdKey= -1, Option= 0
ActCCMessage Already in Table: Code= c006, Mode= 13, Level=  1, CmdKey= -1, Option= 0
ActCCMessage Already in Table: Code= c007, Mode= 13, Level=  1, CmdKey= -1, Option= 0
ActCCMessage Already in Table: Code= 2282, Mode=  0, Level=  0, CmdKey= 8, Option= 0
PnlMsgActionStringAdapter Already in Table: Code= 615e, Mode=  0, Level=  0, CmdKey= -1, Option= 0
18.6.6.0007 Linux/Clang x86_64
Main thread starts: 8B02D100
0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | Loaded log config from /home/diegohum/.local/share/DaVinciResolve/configs/log-conf.xml
0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
```
kernel: 6.9.0-0.rc4.20240416git96fca68c4fbf7.38.fc41.x86_64
intel-compute-runtime-24.09.28717.17-1.fc41.src.rpm
[clinfo.txt](https://github.com/intel/compute-runtime/files/15042460/clinfo.txt)

However Resolve was prone to crashes. I couldn't load any h264/aac media files, it crashes.

jncsotk30, 2024-04-30:
> Using this patch on Arch linux + Davinci Resolve + Intel Arc 770 with latest intel-compute-runtime, everything works good but video looks like this; ![mpv-shot0016](https://private-user-images.githubusercontent.com/13213055/321843952-f6458e78-d2b3-449d-9f58-9c0b6b082f54.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzOTg0OTMsIm5iZiI6MTcxNDM5ODE5MywicGF0aCI6Ii8xMzIxMzA1NS8zMjE4NDM5NTItZjY0NThlNzgtZDJiMy00NDlkLTlmNTgtOWMwYjZiMDgyZjU0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI5VDEzNDMxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc2ZWIxYmJlNGVkN2E2MWZmYTFhMTE4YjEzYzJhMjJhZjg2ZjVkYjJkOGNiYWNkZmZiOTNjNjViZmM4MTU5ZTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UpKR4ElmgN3aiorM2sKeSz08rNR-GrfMhKCElgKsIP0)

@vidalinux, what version of compute-runtime were you using? Just curious since I'm also using an Arc based GPU and am running `24.09.28717.17-1`. Sounds like this won't be able to be fixed unless someone with Arc hardware knows how and gets some free time.

vidalinux, 2024-05-23:
> Resolve doesn't start anymore after Fedora update:
> 
> ```
> #!/bin/bash
> export NEOReadDebugKeys=1
> export DisableDeepBind=1
> export OverrideGpuAddressSpace=48
> /opt/resolve/bin/resolve
> ```
> 
> ```
> diegohum@tower:~$ resolve
> ActCCMessage Already in Table: Code= c005, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> ActCCMessage Already in Table: Code= c006, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> ActCCMessage Already in Table: Code= c007, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> ActCCMessage Already in Table: Code= 2282, Mode=  0, Level=  0, CmdKey= 8, Option= 0
> PnlMsgActionStringAdapter Already in Table: Code= 615e, Mode=  0, Level=  0, CmdKey= -1, Option= 0
> 18.6.6.0007 Linux/Clang x86_64
> Main thread starts: 8B02D100
> 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
> 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | Loaded log config from /home/diegohum/.local/share/DaVinciResolve/configs/log-conf.xml
> 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
> ```
> 
> kernel: 6.9.0-0.rc4.20240416git96fca68c4fbf7.38.fc41.x86_64 intel-compute-runtime-24.09.28717.17-1.fc41.src.rpm [clinfo.txt](https://github.com/intel/compute-runtime/files/15042460/clinfo.txt)
> 
> However Resolve was prone to crashes. I couldn't load any h264/aac media files, it crashes.

Same issue here with Archlinux and latest updates!

specialworld83, 2024-05-23:
> Same issue here with Archlinux and latest updates!

Just tested both in fedora 40 and archlinux and it works correctly with the davinci resolve 18.6.6 studio version
[image]

vidalinux, 2024-05-23:
> > > Resolve doesn't start anymore after Fedora update:
> > > ```
> > > #!/bin/bash
> > > export NEOReadDebugKeys=1
> > > export DisableDeepBind=1
> > > export OverrideGpuAddressSpace=48
> > > /opt/resolve/bin/resolve
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > ```
> > > diegohum@tower:~$ resolve
> > > ActCCMessage Already in Table: Code= c005, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> > > ActCCMessage Already in Table: Code= c006, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> > > ActCCMessage Already in Table: Code= c007, Mode= 13, Level=  1, CmdKey= -1, Option= 0
> > > ActCCMessage Already in Table: Code= 2282, Mode=  0, Level=  0, CmdKey= 8, Option= 0
> > > PnlMsgActionStringAdapter Already in Table: Code= 615e, Mode=  0, Level=  0, CmdKey= -1, Option= 0
> > > 18.6.6.0007 Linux/Clang x86_64
> > > Main thread starts: 8B02D100
> > > 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
> > > 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | Loaded log config from /home/diegohum/.local/share/DaVinciResolve/configs/log-conf.xml
> > > 0x7fa38b02d100 | Undefined            | INFO  | 2024-04-19 17:17:37,269 | --------------------------------------------------------------------------------
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > kernel: 6.9.0-0.rc4.20240416git96fca68c4fbf7.38.fc41.x86_64 intel-compute-runtime-24.09.28717.17-1.fc41.src.rpm [clinfo.txt](https://github.com/intel/compute-runtime/files/15042460/clinfo.txt)
> > > However Resolve was prone to crashes. I couldn't load any h264/aac media files, it crashes.
> > 
> > 
> > Same issue here with Archlinux and latest updates!
> 
> Just tested both in fedora 40 and archlinux and it works correctly with the davinci resolve 18.6.6 studio version ![Schermata_20240523_063121](https://private-user-images.githubusercontent.com/9156314/333033382-18442dcd-d34d-49a4-88b3-3f57d5d938d8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Nzg3MzIsIm5iZiI6MTcxNjQ3ODQzMiwicGF0aCI6Ii85MTU2MzE0LzMzMzAzMzM4Mi0xODQ0MmRjZC1kMzRkLTQ5YTQtODhiMy0zZjU3ZDVkOTM4ZDgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjNUMTUzMzUyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmUzZTY1MjI1YmMwMDc5Yzk0MTA4N2Y1YmVhNDMxNzAxOGZkMzQzYjIxYTY5ZmZhNjk1NzJhMjM3NjhiNzU2MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.EKAL7-MfU5C_urG-02DPFk5OVFcNZ86BsqmI4yLDdpg)

Which GPU you're using? I'm using ARC770

smunaut, 2024-05-23:
Well the ARC770 has never worked so no surprise there ...

specialworld83, 2024-05-23:
> Which GPU you're using? I'm using ARC770

I'm using Iris Xe Graphics

vidalinux, 2024-05-23:
I was able to launch davinci-resolve with Arc770 using this QT_QPA_PLATFORM=xcb /opt/resolve/bin/resolve I think the issue is wayland!

vidalinux, 2024-05-23:
> Well the ARC770 has never worked so no surprise there ...

@smunaut the only issue I'm having is the one already discribed on my previous post about video distortion inside davinci resolve video previews.

 Where we can send you an ARC gpu? the community needs support for ARC gpus! 😃

digitalinferno, 2024-07-09:
> @vidalinux I assume you're on intel Arc ? If so, yeah there is a bug ... but I don't have access to Arc hardware so I can't debug/fix it.

Could you post some instructions on how to debug this for you? This way, we can try to pass some useful data to fix the bug.

smunaut, 2024-07-09:
@digitalinferno Already did. See this branch : https://github.com/smunaut/compute-runtime/tree/fix-arc

I didn't submit it upstream yet because there are some MR from mesa that is pending that will require some changes on the compute runtime side too, so I'm waiting for that to be merged to submit it all at once (so I don't have to go through the whole process twice ...).

But in the mean time, assuming you have mesa >= 24.1 , you can use the above branch and it should work.

eero-t, 2024-07-09:
Intel Developer Cloud seems to have Arc: https://www.intel.com/content/www/us/en/developer/tools/devcloud/edge/hardware-workloads.html

I'm not sure dev cloud is useful for checking anything with a GUI like DaVinci though, I think it's more for testing server (headless compute) workloads...

smunaut, 2024-07-09:
@eero-t Yeah, I had a look at that at some point but :

(1) Seems like you have to pay (kind of make sense), and although I don't mind spending a bit of time to debug something, I'm not gonna spend actual money to fix a problem I don't have myself.

(2) As you said, seemed targeted at  compute workload. Here I need an actual OpenGL context and you can't start X without a screen attached, you'd need to mess with faking EDID or something to get it to send a signal to nothing ... lots of trouble I didn't want to go through just to start debugging ...

digitalinferno, 2024-07-15:
@smunaut compiled `releases/24.26` with your patch (and warnings being treated as errors disabled), launched Resolve with:
```
 #!/bin/bash
export OCL_ICD_VENDORS=/etc/OpenCL/vendors/
export LD_LIBRARY_PATH=/opt/resolve/lib64
export NEOReadDebugKeys=1
export DisableDeepBind=1
export OverrideGpuAddressSpace=48
LD_PRELOAD=/usr/lib64/libglib-2.0.so /opt/resolve/bin/resolve
```
but only Fusion work correctly.

[34 more replies]