intel/compute-runtime #663: clinfo on WSL2 using A770 causes BSOD

maleadt, 2023-07-20:
I'm using the following set-up:

- Windows 10 22H2 (also tested 22H1)
- 64-bit x86 (i5-6600K) with an Arc A770 (ReBAR not enabled/supported on this system, but Above 4G decoding is enabled)
- Latest beta host driver (31.0.101.4575, but also tested the latest WHQL)
- freshly set-up WSL2 Ubuntu (kernel  5.15.90.1-microsoft-standard-WSL2)
- lateste compute-runtime (23.22.26516.18; using the binaries from GitHub)

Doing `clinfo` in a WSL2 terminal starts printing some output, but quickly triggers a BSOD that mentions `dxgmms2.sys` and `SYSTEM_THREAD_EXCEPTION_NOT_HANDLED`. The generated dump file is corrupt, so I couldn't inspect it.

I'm also encountering this BSOD when loading oneAPI.jl, presumably when the first call to Level Zero happens (i.e. `zeInit`).

JablonskiMateusz, 2023-07-20:
Hi @maleadt  could you please ensure that if you remove intel.icd from `/etc/OpenCL/vendors` in the WSL and run `clinfo`, then there is no BSOD? I would like to confirm that the BSOD is related to our package

maleadt, 2023-07-20:
Yes, when I hadn't installed an Intel ICD (i.e. before installing any of the compute-runtime packages) `clinfo` just returned `0 platforms`.

JablonskiMateusz, 2023-07-20:
Thanks for confirmation.

> freshly set-up WSL2 Ubuntu  

which Ubuntu version?

eero-t, 2024-02-28:
Do you mean that there's **Windows** kernel BSOD when trivial operations are done with Linux user-space compute stack under WSL?   (Does not really sound like compute-runtime problem)

maleadt, 2024-02-28:
> Do you mean that there's **Windows** kernel BSOD when trivial operations are done with Linux user-space compute stack under WSL?

Correct.

eero-t, 2024-02-28:
Unless you're using PCI passthrough for the device, and mention of "dgx" (directX) is just co-incidence, I do not see how this could be Linux side compute-runtime problem, rather than Windows driver issue.  Have you reported the issue to Windows side?

Virtual machine using virtualized host drivers should not be able to BSOD the host...

maleadt, 2024-03-11:
After recently updating both NEO/IGC, this issue doesn't occur anymore. Sadly, the native Intel GPU driver was updated too, so it's impossible to tell which updated fixed the issue... Anyway, I'm glad it got fixed, so this can be closed.

closed 2024-03-11 by maleadt

eero-t, 2024-03-11:
Thanks for testing, reporting the results here!