intel/compute-runtime #665: Cannot detect a NEO-supported device if there is another non-NEO-supported Intel device on the system

agrn, 2023-07-27:
Hi,

I have a pretty peculiar system, with a i5-4690k and its associated iGPU (HD Graphics 4600), and an Arc A770.  The iGPU is not supported by the NEO driver (instead, one should use the legacy `intel-ocl` or `beignet` drivers).

With the iGPU enabled, `clinfo` and other programs were not able to detect the Arc device, and `clGetPlatformIDs()` returned `-1001` (`CL_PLATFORM_NOT_FOUND_KHR`).

`strace`ing `clinfo` revealed that the correct ICD was loaded, and tried to access to two peripherals in `/dev/dri`, performed 2 `ioctl()`s, and returned.

Disabling the iGPU in the system firmware (BIOS/UEFI) solved the issue.

This workaround satisfies me, but I can try to fix it if you accept contributions.  I wanted to post this mostly because I did not find any info on this specific issue.

I will post the `strace` output as soon as possible.

agrn, 2023-07-27:
Also, perhaps issue #663 is related?

JablonskiMateusz, 2023-07-27:
hi @agrn 
could you please share strace log ?

agrn, 2023-07-27:
Sorry for the late answer.

Here is the log of `strace` with the iGPU enabled: [clinfo-a770-hd4600.txt](https://github.com/intel/compute-runtime/files/12187483/clinfo-a770-hd4600.txt)

(Sorry for the `openat()` noise, that's a side-effect of NixOS…)

`libigdrcl.so` is opened on line 70, and I believe the actual operation of the ICD begins around line 170.  Both peripherals are opened on line 237 and 239.  There is more `ioctl()`s than I remembered, though.  At the end, you can see that there is no listing of OpenCL capabilities (ie. no platform were reported).

For contrast, here is the log of `strace` with the iGPU disabled: [clinfo-a770-only.txt](https://github.com/intel/compute-runtime/files/12187519/clinfo-a770-only.txt)  

You can see `libigdrcl.so` opened on line 70.  A single device is opened this time, still on line 237.

JablonskiMateusz, 2024-03-08:
@agrn  could you verify if the issue is visible on most recent release?

It should be fixed in f63dd1f4f2648f8becdbc0fbdf455379b5810ee5

agrn, 2024-03-17:
Hi, I'm afraid I no longer have this system, and hence cannot test this change for now.

closed 2024-09-20 by JablonskiMateusz