intel/compute-runtime #669: build error with gcc13

tjaalton, 2023-08-11:
Building with current Debian unstable (which defaults to gcc13) fails:

```
/usr/include/c++/13/bits/stl_algobase.h:398:11: error: ‘*(unsigned char (*)[7])((char*)&<unnamed> + offsetof(NEO::ArgDescValue, NEO::ArgDescValue::elements.StackVec<NEO::ArgDescValue::Element, 1, unsigned char>::onStackMemRawBytes[0]))’ may be used uninitialized [-Werror=maybe-uninitialized]
  398 |         { *__to = *__from; }
      |           ^
```
etc

[build.txt](https://github.com/intel/compute-runtime/files/12320168/build.txt)

k1gen, 2023-08-16:
same, can't compile on Arch with gcc13: https://paste.rs/G3Jhh.txt

alheinecke, 2023-08-18:
have similar issues  on Fedora 38 with default system compiler

matthew-olson-intel, 2023-08-18:
I get the same error using GCC 13 on Arch Linux.

Installing the `gcc-12` package and using the following cmake command works:

```
cmake -DCMAKE_BUILD_TYPE=Release -DNEO_SKIP_UNIT_TESTS=1 -DCMAKE_CXX_COMPILER=g++-12
-DCMAKE_C_COMPILER=gcc-12 ..
```

JablonskiMateusz, 2023-08-23:
Hi folks, 
thanks for reporting the issue. Could you try to use https://github.com/intel/compute-runtime/commit/8b09bb855e2503e033f684da9b1fcd08802c2156 when building with gcc13? I recently added some fixes related to gcc13 compilation

tjaalton, 2023-08-23:
I still get the same error

JablonskiMateusz, 2023-08-23:
the initial error should gone after https://github.com/intel/compute-runtime/commit/894bab05efb447d249c9f0afdbf0ca62c82395c6

tjaalton, 2023-08-23:
cherry-picked on top of 23.26.26690.22

k1gen, 2023-08-24:
> Hi folks, thanks for reporting the issue. Could you try to use [8b09bb8](https://github.com/intel/compute-runtime/commit/8b09bb855e2503e033f684da9b1fcd08802c2156) when building with gcc13? I recently added some fixes related to gcc13 compilation

fixed for me on Arch, thanks!

tjaalton, 2023-09-15:
So there's something else missing from the current release branches that still make it fail, any idea what those might be? Building current master succeeds.

JablonskiMateusz, 2023-09-18:
@kmachnik please provide ETA

tjaalton, 2023-10-03:
23.35 builds fine

closed 2023-10-03 by tjaalton