intel/compute-runtime #671: [OpenCL] Support for __opencl_c_device_enqueue

sudden6, 2023-08-19:
The old driver supported device side kernel enqueue on my Kaby Lake iGPU, are there any plans to support this with the OpenCL NEO stack?

sudden6, 2023-08-25:
Hi @JablonskiMateusz thank you for your answer.

Just to clarify, before I close this, does that mean it will *never* be supported again? And just out curiosity, what were the reasons why it was dropped?

AdamCetnerowski, 2023-09-04:
Correct, there is no intention to re-implement this. 

Device Enqueue was moved to optional in OpenCL 3.0 spec. This is the most complex feature in the standard, requires a lot of effort to implement & maintain, while the ecosystem adoption was almost nonexistent.