intel/compute-runtime #674: WSL error on os_interface.h

Serizao, 2023-09-01:
Hello, I have an error when using torch with WSL. :
```
No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda-12/'
Abort was called at 62 line in file:
../../neo/shared/source/os_interface/os_interface.h
Aborted
````
I don't think the problem comes from CUDA because it is correctly installed.

-> wsl : unbuntu 22.04
-> windows 11 pro
-> GPU : ARC A770
-> cpu: i7 12700

What do you need to investigate further?

JablonskiMateusz, 2023-09-04:
could you share callstack?

eero-t, 2024-02-28:
What "clinfo" tells, does it detect GPU?

Are you using PCI pass-through from host for the GPU device, or something like DXGKRN?   Have you installed both (new enough) kernel and user-space GPU drivers to the Ubuntu setup?