intel/compute-runtime #677: Macro conflict/redefinition between compute-runtime and opencl-headers

roysjosh, 2023-09-24:
Attempting to compile the latest compute-runtime on a vanilla Fedora 38 led to a compile failure complaining of a macro redefinition: `CL_MEM_DEVICE_ID_INTEL`. This was introduced in 1f21d34fd37e58aa24b699a79621e3e6ee9c779f in this repo and https://github.com/KhronosGroup/OpenCL-Headers/commit/7bcc0f18ab837a446360f4c17f245102429508bf a few months ago. I didn't notice this in F37 which is stuck on opencl-headers 3.0-12.20220510.

I've replaced all occurrences of the macro in this repo, definition and use, which allows me to finish the build.

JablonskiMateusz, 2023-09-27:
Hi @roysjosh ,
Thanks for reporting the issue. `CL_MEM_DEVICE_ID_INTEL` was added as experimental feature but since it is available in official headers I hope we can update headers. However, we found that the most recent release of headers introduced unexpected mismatch in VME specific enum type which is already fixed on top of repo (https://github.com/KhronosGroup/OpenCL-Headers/pull/241). We need to wait for next release tag.

JablonskiMateusz, 2024-01-29:
updated headers in https://github.com/intel/compute-runtime/commit/327d4c4d48c736e8db1a54db54cbcd5bb4b27d8b

closed 2024-03-08 by JablonskiMateusz