intel/compute-runtime #679: Gemini Lake clpeak fails on Ubuntu 22.04 latest kernel

looi, 2023-09-28:
System: Dell Wyse 5070 Intel Celeron J4105 (Gemini Lake)
Intel Compute Runtime [23.30.26918.9](https://github.com/intel/compute-runtime/releases/tag/23.30.26918.9) installed with official instructions.

## What works

* Ubuntu 20.04 kernel 5.4: Works out of the box
* Ubuntu 22.04 kernel 5.15: Doesn't work out of the box, but works with `i915.enable_hangcheck=0 i915.request_timeout_ms=100000`, see https://github.com/intel/compute-runtime/issues/497
* Without the above params, clpeak hangs with the following kernel logs:
```
[  101.920695] Fence expiration time out i915-0000:00:02.0:clpeak[969]:454!
[  101.920754] Fence expiration time out i915-0000:00:02.0:clpeak[969]:452!
[  101.920760] Fence expiration time out i915-0000:00:02.0:clpeak[969]:450!
```

## What doesn't work

- Ubuntu 22.04 kernel 6.2
- clpeak fails, and produces error clFinish (-5) with kernel logs (note that these logs were never seen with kernel 5.15):
```
[   70.897984] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[   70.898106] i915 0000:00:02.0: [drm] clpeak[910] context reset due to GPU hang
[   70.907884] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:e757fefe, in clpeak [910]
...
[  170.596540] i915 0000:00:02.0: [drm] Resetting rcs0 for CS error
[  170.596659] i915 0000:00:02.0: [drm] clpeak[910] context reset due to GPU hang
[  170.606125] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:ac045407, in clpeak [910]
```
- Setting the params that fixed 5.15 (`i915.enable_hangcheck=0 i915.request_timeout_ms=100000`) or increasing to 200000 doesn't appear to help.
- Vulkan compute (vkpeak) works fine. So this appears to be a problem specific to Intel Compute Runtime.

Ar-Ray-code, 2023-12-08:
I have the same problem on both Celeron J4125 and N4000.
I am seeing the following error in GPU inference in OpenVINO 2023.1.0.

```bash
terminate called after throwing an instance of 'InferenceEngine::GeneralError'
  what(): [ GENERAL_ERROR ] Check 'false' failed at src/plugins/intel_gpu/src/plugin/program.cpp:401:
GPU program build failed!
[GPU] clWaitForEvents, error code: -14
````

Note that Linux kernel 5.15 does not have the problem.

nyanmisaka, 2023-12-08:
I heard about this issue a year ago but I no longer have a GLK device now.
This is a kernel regression because Gemini Lake/GLK only fails when using the new kernel.

@looi @Ar-Ray-code
Better to file an issue in drm/intel. https://gitlab.freedesktop.org/drm/intel/-/issues/?label_name%5B%5D=Community

looi, 2023-12-16:
I don't think this is necessarily a kernel regression, because as I have stated above, vulkan compute works fine.

Personally, I have switched to using vulkan. The performance is comparable (especially making proper use of vulkan subgroups), but more importantly, it seems to be much more stable on both Windows and Linux. Intel Compute Runtime / OpenCL has weird issues like this one. Vulkan also seems to work much better on non-intel GPUs, especially nvidia, where they refuse to support basic features like subgroups and half-precision floats in OpenCL. So I feel like vulkan is the future and OpenCL is dying anyways.

nyanmisaka, 2023-12-16:
> What works
> Ubuntu 22.04 kernel 5.15: Doesn't work out of the box, but works with i915.enable_hangcheck=0

> What doesn't work
> Ubuntu 22.04 kernel 6.2

Your input suggests this is a kernel regression. The only difference between whether it works or not is the kernel version, bisect the commit between the two should find the culprit.

This isn't the first time I've seen i915 regression, [last time](https://gitlab.freedesktop.org/drm/intel/-/issues/7627) it even broke both the Vulkan compute and OpenCL.

looi, 2023-12-16:
I agree that a kernel change broke Intel Compute Runtime. I guess whether or not it's a kernel regression is a subjective question depending on what exactly caused the breakage. Maybe Intel Compute Runtime is making incorrect assumptions about i915 or relying on undefined behavior, in which case it would not be a kernel regression. Given that vulkan compute still works, I think it is a likely possibility.

Suxsem, 2024-08-24:
any news on this one? same issue

Suxsem, 2024-08-27:
@JablonskiMateusz any chance you could take a look? thank you

quantumint, 2024-10-09:
I have the same issue with J5005 Gemini Lake and  Proxmox 8.2.7 running kernel  6.8.12-2-pve and the most recent Intel Compute Runtime [24.35.30872.22](https://github.com/intel/compute-runtime/releases/tag/24.35.30872.22)
It prevents OpenVINO from working with "[GPU] clWaitForEvents, error code: -14" error. It seems that recent releases did not fix the problem. 

Old kernel 5.15.108-1-pve works out of the box.

aeozyalcin, 2024-11-02:
I am also having the same issue on my J5005 running Ubuntu 24 LTS with kernel 6.8.0-48 with the updated OpenVINO runtime. Not able to use the GPU for acceleration. This was working fine on older kernels.

Fossan, 2025-07-31:
Anyone managed to run the OpenVINO on J5005?
Proxmox: `8.4.1`
Kernel: `6.8.12-11-pve`
OpenVINO: `2025.3.0-19620-2739b894c21` (nightly build)
intel-compute-runtime: `24.35.30872.22`

```
(openvino_nightly_env) root@frigate:/# benchmark_app -m /openvino-model/ssdlite_mobilenet_v2.xml -d GPU
[Step 1/11] Parsing and validating input arguments
[ INFO ] Parsing input parameters
[Step 2/11] Loading OpenVINO Runtime
[ INFO ] OpenVINO:
[ INFO ] Build ................................. 2025.3.0-19620-2739b894c21
[ INFO ] 
[ INFO ] Device info:
[ INFO ] GPU
[ INFO ] Build ................................. 2025.3.0-19620-2739b894c21
[ INFO ] 
[ INFO ] 
[Step 3/11] Setting device configuration
[ WARNING ] Performance hint was not explicitly specified in command line. Device(GPU) performance hint will be set to PerformanceMode.THROUGHPUT.
[Step 4/11] Reading model files
[ INFO ] Loading model files
[ INFO ] Read model took 67.02 ms
[ INFO ] Original model I/O parameters:
[ INFO ] Model inputs:
[ INFO ]     image_tensor:0 (node: image_tensor) : u8 / [...] / [1,300,300,3]
[ INFO ] Model outputs:
[ INFO ]     detection_boxes:0 (node: DetectionOutput) : f32 / [...] / [1,1,100,7]
[Step 5/11] Resizing model to match image sizes and given batch
[ INFO ] Model batch size: 1
[Step 6/11] Configuring input of the model
[ INFO ] Model inputs:
[ INFO ]     image_tensor:0 (node: image_tensor) : u8 / [N,H,W,C] / [1,300,300,3]
[ INFO ] Model outputs:
[ INFO ]     detection_boxes:0 (node: DetectionOutput) : f32 / [...] / [1,1,100,7]
[Step 7/11] Loading the model to the device
[ ERROR ] Exception from src/inference/src/cpp/core.cpp:109:
Exception from src/inference/src/dev/plugin.cpp:53:
Check 'false' failed at src/plugins/intel_gpu/src/plugin/program_builder.cpp:163:
[GPU] ProgramBuilder build failed!
Exception from src/plugins/intel_gpu/src/runtime/ocl/ocl_memory.cpp:142:
[GPU] clWaitForEvents, error code: -14 CL_EXEC_STATUS_ERROR_FOR_EVENTS_IN_WAIT_LIST

Traceback (most recent call last):
  File "/openvino_nightly_env/lib/python3.9/site-packages/openvino/tools/benchmark/main.py", line 449, in main
    compiled_model = benchmark.core.compile_model(model, benchmark.device, device_config)
  File "/openvino_nightly_env/lib/python3.9/site-packages/openvino/_ov_api.py", line 599, in compile_model
    super().compile_model(model, device_name, {} if config is None else config),
RuntimeError: Exception from src/inference/src/cpp/core.cpp:109:
Exception from src/inference/src/dev/plugin.cpp:53:
Check 'false' failed at src/plugins/intel_gpu/src/plugin/program_builder.cpp:163:
[GPU] ProgramBuilder build failed!
Exception from src/plugins/intel_gpu/src/runtime/ocl/ocl_memory.cpp:142:
[GPU] clWaitForEvents, error code: -14 CL_EXEC_STATUS_ERROR_FOR_EVENTS_IN_WAIT_LIST
```

JablonskiMateusz, 2025-08-05:
@Fossan  could you retest using most recent release from 24.35 branch (https://github.com/intel/compute-runtime/releases/tag/24.35.30872.36) ?

Fossan, 2025-08-13:
> [@Fossan](https://github.com/Fossan) could you retest using most recent release from 24.35 branch (https://github.com/intel/compute-runtime/releases/tag/24.35.30872.36) ?

I have rolled back the kernel to 5.15 (which solved the issue by the way), but will give it a try with a newer kernel in few days and keep you posted here. Thanks for the heads up!

aeozyalcin, 2025-08-24:
> [@Fossan](https://github.com/Fossan) could you retest using most recent release from 24.35 branch (https://github.com/intel/compute-runtime/releases/tag/24.35.30872.36) ?

I have had success with OpenVINO GPU acceleration on Kernel 6.15.11-061511-generic with runtime 24.35.30872.36! Well done! @Fossan

Fossan, 2025-08-26:
> [@Fossan](https://github.com/Fossan) could you retest using most recent release from 24.35 branch (https://github.com/intel/compute-runtime/releases/tag/24.35.30872.36) ?

Took me a while to replace the runtime in the container, but it seems to be working well on `6.8.12-11-pve` kernel now:

[image]

```
root@11cef0a17e65:/# benchmark_app -m /openvino-model/ssdlite_mobilenet_v2.xml -d GPU
[Step 1/11] Parsing and validating input arguments
[ INFO ] Parsing input parameters
[Step 2/11] Loading OpenVINO Runtime
[ INFO ] OpenVINO:
[ INFO ] Build ................................. 2024.4.0-16579-c3152d32c9c-releases/2024/4
[ INFO ] 
[ INFO ] Device info:
[ INFO ] GPU
[ INFO ] Build ................................. 2024.4.0-16579-c3152d32c9c-releases/2024/4
[ INFO ] 
[ INFO ] 
[Step 3/11] Setting device configuration
[ WARNING ] Performance hint was not explicitly specified in command line. Device(GPU) performance hint will be set to PerformanceMode.THROUGHPUT.
[Step 4/11] Reading model files
[ INFO ] Loading model files
[ INFO ] Read model took 62.30 ms
[ INFO ] Original model I/O parameters:
[ INFO ] Model inputs:
[ INFO ]     image_tensor:0 (node: image_tensor) : u8 / [...] / [1,300,300,3]
[ INFO ] Model outputs:
[ INFO ]     detection_boxes:0 (node: DetectionOutput) : f32 / [...] / [1,1,100,7]
[Step 5/11] Resizing model to match image sizes and given batch
[ INFO ] Model batch size: 1
[Step 6/11] Configuring input of the model
[ INFO ] Model inputs:
[ INFO ]     image_tensor:0 (node: image_tensor) : u8 / [N,H,W,C] / [1,300,300,3]
[ INFO ] Model outputs:
[ INFO ]     detection_boxes:0 (node: DetectionOutput) : f32 / [...] / [1,1,100,7]
[Step 7/11] Loading the model to the device
[ INFO ] Compile model took 1202.82 ms
[Step 8/11] Querying optimal runtime parameters
[ INFO ] Model:
[ INFO ]   NETWORK_NAME: frozen_inference_graph
[ INFO ]   OPTIMAL_NUMBER_OF_INFER_REQUESTS: 4
[ INFO ]   PERF_COUNT: False
[ INFO ]   ENABLE_CPU_PINNING: False
[ INFO ]   MODEL_PRIORITY: Priority.MEDIUM
[ INFO ]   GPU_HOST_TASK_PRIORITY: Priority.MEDIUM
[ INFO ]   GPU_QUEUE_PRIORITY: Priority.MEDIUM
[ INFO ]   GPU_QUEUE_THROTTLE: Priority.MEDIUM
[ INFO ]   GPU_ENABLE_LOOP_UNROLLING: True
[ INFO ]   GPU_DISABLE_WINOGRAD_CONVOLUTION: False
[ INFO ]   CACHE_DIR: 
[ INFO ]   CACHE_MODE: CacheMode.OPTIMIZE_SPEED
[ INFO ]   PERFORMANCE_HINT: PerformanceMode.THROUGHPUT
[ INFO ]   EXECUTION_MODE_HINT: ExecutionMode.PERFORMANCE
[ INFO ]   COMPILATION_NUM_THREADS: 4
[ INFO ]   NUM_STREAMS: 2
[ INFO ]   PERFORMANCE_HINT_NUM_REQUESTS: 0
[ INFO ]   INFERENCE_PRECISION_HINT: <Type: 'float16'>
[ INFO ]   DYNAMIC_QUANTIZATION_GROUP_SIZE: 0
[ INFO ]   DEVICE_ID: 0
[ INFO ]   EXECUTION_DEVICES: ['GPU.0']
[Step 9/11] Creating infer requests and preparing input tensors
[ WARNING ] No input files were given for input 'image_tensor:0'!. This input will be filled with random values!
[ INFO ] Fill input 'image_tensor:0' with random values 
[Step 10/11] Measuring performance (Start inference asynchronously, 4 inference requests, limits: 60000 ms duration)
[ INFO ] Benchmarking in inference only mode (inputs filling are not included in measurement loop).
[ INFO ] First inference took 50.21 ms
[Step 11/11] Dumping statistics report
[ INFO ] Execution Devices:['GPU.0']
[ INFO ] Count:            2060 iterations
[ INFO ] Duration:         60186.18 ms
[ INFO ] Latency:
[ INFO ]    Median:        113.64 ms
[ INFO ]    Average:       116.05 ms
[ INFO ]    Min:           70.30 ms
[ INFO ]    Max:           185.85 ms
[ INFO ] Throughput:   34.23 FPS
```

Thanks for your effort @JablonskiMateusz!

karentsmells1, 2025-09-25:
@Fossan  - Any chance you can post your config files for docker compose and the frigate config? I am trying to get this to work on Proxmox and am pulling my hair out. I pinned to the kernel you mentioned above and can't get it to work. Hoping to not have to buy a coral TPU or pin to the old 5.15 kernel.

rufciocz, 2025-10-31:
Hello,﻿ @pwilma ﻿ 
Thank you for the update. I can confirm that Frigate is now functioning correctly. I upgraded the host kernel to version 6.8.12-9-pve and created a new container with Ubuntu 22.04. Following the recompose of the Frigate stack in Portainer as proposed by @sumitc96, no errors are present, and the system operates smoothly and reliably.

aeozyalcin, 2025-11-09:
> Hi @looi , @rufciocz ,
> We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

Hi @pwilma . Can we expect the new releases going forward just working with these Intel CPUs now, or was this a one off release to extend support?