intel/compute-runtime #681: Zombie processes

pvelesko, 2023-09-30:
Is there a way to reload/reset the driver to get rid of these dead processes? These are a result of running various Level Zero tests overnight. 

[image]

JablonskiMateusz, 2023-10-06:
Hi @pvelesko 
What are these processes, who spawned them?

pvelesko, 2023-10-06:
We have a large set of tests that we run on the Level Zero backend. Some of these tests produce these processes which I can't figure out how to kill. I tried unbinding dgpu from the i915 driver, I tried kill -9, but so far the only way I've found is to :
```
alias reboot_hardcore="history -a && sudo sh -c 'echo b > /proc/sysrq-trigger'"
```

A regular reboot doesn't work either as the system just hangs while waiting on something. Is there a way to reset the driver completely? Since we can't unload the driver while it's in use.

HoppeMateusz, 2023-10-06:
Hello,

have you tried unbinding i915 like this ?

sudo sh -c "echo -n auto > /sys/bus/pci/devices/0000\:00\:02.0/power/control"
sudo sh -c "echo -n "0000:00:02.0" > /sys/bus/pci/drivers/i915/unbind"
sudo modprobe -r i915

PCI number can be found with:

lspci | grep -i display 
or
lspci | grep -i vga

pvelesko, 2023-10-06:
I'll try this, thank you.

pvelesko, 2023-10-07:
`modprobe: FATAL: Module i915 is in use.` after unbinding devices.

HoppeMateusz, 2023-10-09:
hello, 
thanks for a try, 
Unloading i915 might not work if KMD is hanging somewhere, reboot should be the ultimate way of resetting.

Can you provide details for reproducing the issue?
What platform is it?
OS version
i915 version
possibly a reproducer
Thanks

pwilma, 2025-10-29:
Hi @pvelesko ,
We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!