intel/compute-runtime #682: Build issues: `Error! Loading of FCL library has failed! Filename: libigdfcl.so.1`

pvelesko, 2023-10-01:
```
[ 59%] Generating ../../../../bin/built_ins/x64/spirv/fill_image3d.spv
Error! Loading of FCL library has failed! Filename: libigdfcl.so.1
Error! FCL initialization failure. Error code = -6
Command was: /home/pvelesko/intel-compute-runtime-build/neo/build/bin/ocloc -q -spv_only -file /home/pvelesko/intel-compute-runtime-build/neo/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /home/pvelesko/intel-compute-runtime-build/neo/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
make[2]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make:220: bin/built_ins/x64/spirv/fill_image3d.spv] Error 250
make[1]: *** [CMakeFiles/Makefile2:38628: shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

JablonskiMateusz, 2023-10-05:
Hi @pvelesko
it seems you have no igc installed on the system. You need IGC/GMM/Level zero dependencies based on release notes

pvelesko, 2023-10-05:
This is part of my install script: https://github.com/pvelesko/intel-compute-runtime-build @JablonskiMateusz 

I guess I'm missing -DIGC_DIR=?

JablonskiMateusz, 2023-10-06:
I'm not expert in building from source but I hope that if you build and install IGC, you should have both libigc and libigdfcl libraries installed on the system

pvelesko, 2023-10-06:
I'm trying to install it on a shared system where I don't have root

JablonskiMateusz, 2023-10-12:
you could try to build igc packages and then extract them to location where you have access and then somehow point them so pkg-config finds it on cmake generation when building neo

pvelesko, 2023-10-12:
After install, level zero doesn't detect a device. Are the any debug options to check if the kernel space driver installed works for the neo release? @JablonskiMateusz

pvelesko, 2024-01-16:
@JablonskiMateusz  ping @pengtu

pvelesko, 2024-01-17:
I see a similar issue has been open since 2020

https://github.com/intel/compute-runtime/issues/265

pvelesko, 2024-01-18:
I was ultimately able to resolve the issue.. Turns out that LLVM must use static linking!...  I'll keep this issue open until this is either fixed or documentation is updated to reflect this requirement.  @JablonskiMateusz @pengtu

libmld-dev, 2024-01-25:
I had a similar issue with the same error. My problem was, that the FCL Library was using the SPIRV-LLVM-Translator for llvm13.

In the [igc build description](https://github.com/intel/intel-graphics-compiler/blob/master/documentation/build_ubuntu.md#use-preinstalled-packages) the package libllvmspirvlib-dev is suggested to be installed.

However, on Ubuntu 22.04.3 this will install libllvmspirvlib13, which wont work if you build with prebuild llvm14. The fix was installing libllvmspirvlib-14-dev instead.

pvelesko, 2024-01-25:
I ran into similar issues as yours as well so I ended up compiling everything from source @libmld-dev  but that revealed the issue described here.

pvelesko, 2024-01-25:
>  to resolve the issue.. Turns out that LLVM must use static linking!..

> From compute runtime point of view I don't see we could fix it.
Could you please open an issue in https://github.com/intel/intel-graphics-compiler/ and continue discussion there?

intel graphics compiler built successfully even with dynamically linked LLVM.

@JablonskiMateusz

chsasank, 2024-02-25:
Hey, @pvelesko 

I am facing the same issue when building compute runtime. How did you fix the issue? Can you share the exact commands?

pvelesko, 2024-02-25:
@chsasank I recompiled LLVM and used static linking which resolved this issue. This should be noted somewhere I think @JablonskiMateusz  

@chsasank You can use my configure and build scripts https://github.com/pvelesko/intel-compute-runtime-build

chsasank, 2024-02-25:
Hey, I have gone through your scripts and have not been able to find the exact line for static linking for LLVM.

pvelesko, 2024-02-25:
@chsasank 

https://github.com/pvelesko/intel-compute-runtime-build/blob/main/configure-llvm.sh 

this will configure a static build (default for llvm)

chsasank, 2024-02-25:
Got it. [Default](https://llvm.org/docs/CMake.html) for BUILD_SHARED_LIBS is indeed off. I built everything from source as well using instructions from [here](https://github.com/intel/intel-graphics-compiler/blob/master/documentation/build_ubuntu.md#build-from-sources). That builds LLVM as well - so the build script should've explicitly turned on the `BUILD_SHARED_LIBS`. If can fix it, I'll contribute the PR to that repo.

pvelesko, 2024-02-25:
Building LLVM in-tree using those instructions has worked for me in the past, I believe.

chsasank, 2024-02-25:
The error image itself is actually coming from this repo. See this line: https://github.com/intel/compute-runtime/blob/3341de7a0d5fddd2ea5f505b5d2ef5c13faa0681/shared/offline_compiler/source/offline_compiler.cpp#L692

chsasank, 2024-02-25:
I don't think this error is related to LLVM static linking. Seems to be a dependency issue.

pvelesko, 2024-02-25:
I've probably spent over 20 hours trying to build this thing and switching to static linking LLVM was what made it work. Perhaps you can try building via the script that I sent?

pvelesko, 2024-03-06:
@chsasank  any luck?

chsasank, 2024-03-06:
I have stopped working on this for time being. Will report to you if I succeed later.