intel/compute-runtime #684: Incorrect ZE API versioning for 23.30 release

abagusetty, 2023-10-05:
Reported here: https://github.com/oneapi-src/level-zero/issues/70#issuecomment-1749631610

It seems like the API versioning likely needs to be updated to 1.7 for 23.30 release. 
Issue: https://github.com/intel/compute-runtime/blob/0caf52bd4e96373d6de4026b5f5b29189a1fc484/level_zero/core/source/driver/driver_handle_imp.cpp#L105

Tagging @jandres742  for reach

JablonskiMateusz, 2023-10-06:
Hi @abagusetty 
L0 loader dependency affects building project as newer loader may add some new API. The reported version ensures user that given L0 version is fully supported.

abagusetty, 2023-10-09:
@JablonskiMateusz thanks for the info. Not sure if this is the right place or L0 spec/loader to ask this question: 

If I want to implement feature that are L0-spec version dependent: Is there a way to query for L0-spec version; Given that `zeDriverGetProperties` always seems to return 1_3.

jandres742, 2023-11-09:
> @JablonskiMateusz thanks for the info. Not sure if this is the right place or L0 spec/loader to ask this question:
> 
> If I want to implement feature that are L0-spec version dependent: Is there a way to query for L0-spec version; Given that `zeDriverGetProperties` always seems to return 1_3.

hi @abagusetty . if you mean that you would like to use a feature that is defined in a specification version after v1.3, how do you know if it is supported in the driver? no, L0 doesnt have that. That is, we would need a per-API query to know if a new function or struct defined in 1.7 for instance is already available in the L0 driver, even though L0 driver is reporting v1.3.

as @JablonskiMateusz  mentions, 1.3 means that L0 driver is stating that fully supports at least 1.3. Functions and structs of later versions may be supported, but the version in the driver cannot be upgraded until the L0 driver is fully spec compliant with it.

So in that case, recommendation is either check code or just try to use the function or struct you need, and see if the driver returns an error (like ZE_RESULT_ERROR_UNSUPPORTED_FEATURE) or if it finds the entry in the function tables.

Please let us know if that helps.

abagusetty, 2023-11-09:
@jandres742 Thanks for clarifying. Yes, I was earlier looking to add support for `ZE_COMMAND_LIST_FLAG_IN_ORDER` before your issue (https://github.com/oneapi-src/unified-runtime/issues/941) which was only available from L0-spec 1.7.0 onwards, so just wanted to see if a specific feature is supported by querying the spec-version.

Closing this thread.

closed 2023-11-09 by abagusetty