intel/compute-runtime #686: zesDeviceEnumMemoryModules() crashed after zesInit()

bgoglin, 2023-10-26:
Hello,

When I initialize Sysman with zesInit() but without setting ZES_ENABLE_SYSMAN=1 in the environment, zesDeviceEnumMemoryModules() crashes. Can be reproduced with the following code:
```
#include <level_zero/zes_api.h>
#include <assert.h>
#include <stdio.h>

int main()
{
  ze_driver_handle_t drh;
  ze_device_handle_t dvh;
  zes_device_handle_t dvs;
  uint32_t n;
  int res;

  res = zesInit(0);
  assert(res == ZE_RESULT_SUCCESS);
  printf("zesInit() ok\n");

  res = zeInit(0);
  assert(res == ZE_RESULT_SUCCESS);
  printf("zeInit() ok\n");

  n = 1;
  res = zeDriverGet(&n, &drh);
  assert(res == ZE_RESULT_SUCCESS);
  assert(n >= 1);
  printf("zeDriverGet() ok\n");

  n = 1;
  res = zeDeviceGet(drh, &n, &dvh);
  assert(res == ZE_RESULT_SUCCESS);
  assert(n >= 1);
  printf("zesDeviceGet() ok\n");

  dvs = (zes_device_handle_t) dvh;
  n = 0;
  res = zesDeviceEnumMemoryModules(dvs, &n, NULL);
  assert(res == ZE_RESULT_SUCCESS);
  printf("zesDeviceEnumMemoryModules() ok\n");

  return 0;
}
```
With ZES_ENABLE_SYSMAN=1, it works fine.

I running these packages (latest release) on a Debian machine with Intel(R) Core(TM) i7-7600U CPU (HD Graphics 620 (rev 02)):

intel-igc-core_1.0.14828.8_amd64.deb
intel-igc-opencl_1.0.14828.8_amd64.deb
intel-level-zero-gpu_1.3.26918.9_amd64.deb
intel-opencl-icd_23.30.26918.9_amd64.deb
level-zero-devel_1.13.1+u22.04_amd64.deb
level-zero_1.13.1+u22.04_amd64.deb
libigdgmm12_22.3.0_amd64.deb

Brice

jitendrasharma1989, 2023-11-09:
@bgoglin 

In order to use sysman initialized with zesInit, could you please follow pseudo code mentioned in spec (https://spec.oneapi.io/level-zero/latest/sysman/PROG.html) :

```
function main( ... )
    if ([zesInit](https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesinit)(0) != [ZE_RESULT_SUCCESS](https://spec.oneapi.io/level-zero/latest/core/api.html#ze-result-t))
        output("Can't initialize the API")
    else
        # Discover all the drivers
        uint32_t driversCount = 0
        [zesDriverGet](https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesdriverget)(&driversCount, nullptr)
        zes_driver_handle_t* allDrivers = allocate(driversCount * sizeof(zes_driver_handle_t))
        [zesDriverGet](https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesdriverget)(&driversCount, allDrivers)

        zes_driver_handle_t hDriver = nullptr
        for(i = 0 .. driversCount-1)
            # Discover devices in a driver
            uint32_t deviceCount = 0
            [zesDeviceGet](https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesdeviceget)(allDrivers[i], &deviceCount, nullptr)

            zes_device_handle_t* hSysmanHandles =
                allocate_memory(deviceCount * sizeof(zes_device_handle_t))
            [zesDeviceGet](https://spec.oneapi.io/level-zero/latest/sysman/api.html#zesdeviceget)(allDrivers[i], &deviceCount, hSysmanHandles)

            # Use the hSymanHandles to manage the devices

    free_memory(...)

```

In your sample code, you are trying to typecast device handles obtained from zeDeviceGet(This method would work with sysman which is initialized with ZES_ENABLE_SYSMAN=1 + zeInit). Instead in order to initialize sysman completely with zesInit, please use below APIs (shown in above pseudo code also):

zesInit
zesDriverGet
zesDeviceGet --> It will give list of sysman handles. And use these sysman handles to interact with sysman APIs.

bgoglin, 2023-11-09:
Is there a way to convert a ZES device to/from a ZE device without casting when using zesInit()?
Your link says "For compatibility, an application may also use the Level0 Core API to enumerate through available accelerator devices in the system. For each device handle, an application can cast it to a sysman device handle to manage the system resources of the device."

jitendrasharma1989, 2023-11-09:
>>> Is there a way to convert a ZES device to/from a ZE device without casting when using zesInit()?
No

>>> Your link says ..
So, its a existing method of sysman initialization. So as you could see in that link(two sample pseudo codes illustrates, sysman initialization with both methods) user has a choice to initialize sysman with either :
1) zesInit + zesDriverGet + zesDeviceGet --> and use sysman handles received from zesDeviceGet in Sysman APIs
or
2) ZES_ENABLE_SYSMAN=1 + zeInit + zeDriverGet + zeDeviceGet --> and use typecasted sysman handle in Sysman APIs

bgoglin, 2023-11-09:
Ok the text between both samples should be clarified then. Right now it doesn't say that casting is only valid with ZES_ENABLE_SYSMAN=1.

Is there a guarantee that driver[X].device[Y] obtained from ZE corresponds to driver[X].device[Y] from ZES? Or is there any risk of these device and driver arrays being ordered differently in ZE and ZES? This should be explicit in the doc of ze{,s}{Driver,Device}Get()

saik-intel, 2023-11-10:
>>Is there a guarantee that driver[X].device[Y] obtained from ZE corresponds to driver[X].device[Y] from ZES? Or is there any risk of these device and driver arrays being ordered differently in ZE and ZES? This should be explicit in the doc of ze{,s}{Driver,Device}Get()
[SaiKishore]  We are enhancing the L0 Spec with new API, to map device handle obtained from ze (basically from L0 Core) to device handle obtained from zes (basically from L0 Sysman), we will update on it when it is available

closed 2024-01-25 by JablonskiMateusz

bgoglin, 2024-03-28:
> > > [SaiKishore]  We are enhancing the L0 Spec with new API, to map device handle obtained from ze (basically from L0 Core) to device handle obtained from zes (basically from L0 Sysman), we will update on it when it is available

@saik-intel Hello, is there any news about the specification and implementation of this new API to map device handle between core and zes ?