intel/compute-runtime #688: Abort was called at 54 line in file: ./shared/source/os_interface/windows/wddm/create_um_km_data_translator.cpp

edmondium, 2023-10-31:
hello,
i get this error when i execute "clinfo" after installing "intel-opencl-icd" from ubuntu repository. any help will be appreciated.

JablonskiMateusz, 2023-11-09:
Hi @edmondium , could you share more details about your setup?

edmondium, 2023-11-09:
> Hi @edmondium , could you share more details about your setup?

edmondium@LAPTOP-1Q9H40K6:~$ uname -a
Linux LAPTOP-1Q9H40K6 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
edmondium@LAPTOP-1Q9H40K6:~$ clinfo
Number of platforms                               0

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.3.2
  ICD loader Profile                              OpenCL 3.0
edmondium@LAPTOP-1Q9H40K6:~$ sudo apt install intel-opencl-icd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libclang-cpp14 libigc1 libigdfcl1 libllvm14 libllvmspirvlib14 libopencl-clang14
The following NEW packages will be installed:
  intel-opencl-icd libclang-cpp14 libigc1 libigdfcl1 libllvm14 libllvmspirvlib14 libopencl-clang14
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 51.3 MB of archives.
After this operation, 242 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libllvm14 amd64 1:14.0.6-16 [24.0 MB]
Get:2 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libllvmspirvlib14 amd64 14.0.0-9 [696 kB]
Get:3 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libigc1 amd64 1.0.12504.6-1 [10.3 MB]
Get:4 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libclang-cpp14 amd64 1:14.0.6-16 [12.2 MB]
Get:5 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libopencl-clang14 amd64 14.0.0-4 [101 kB]
Get:6 http://archive.ubuntu.com/ubuntu mantic/universe amd64 libigdfcl1 amd64 1.0.12504.6-1 [90.8 kB]
Get:7 http://archive.ubuntu.com/ubuntu mantic/universe amd64 intel-opencl-icd amd64 22.43.24595.41-1 [3,923 kB]
Fetched 51.3 MB in 26s (1,946 kB/s)
Selecting previously unselected package libllvm14:amd64.
(Reading database ... 129140 files and directories currently installed.)
Preparing to unpack .../0-libllvm14_1%3a14.0.6-16_amd64.deb ...
Unpacking libllvm14:amd64 (1:14.0.6-16) ...
Selecting previously unselected package libllvmspirvlib14:amd64.
Preparing to unpack .../1-libllvmspirvlib14_14.0.0-9_amd64.deb ...
Unpacking libllvmspirvlib14:amd64 (14.0.0-9) ...
Selecting previously unselected package libigc1:amd64.
Preparing to unpack .../2-libigc1_1.0.12504.6-1_amd64.deb ...
Unpacking libigc1:amd64 (1.0.12504.6-1) ...
Selecting previously unselected package libclang-cpp14.
Preparing to unpack .../3-libclang-cpp14_1%3a14.0.6-16_amd64.deb ...
Unpacking libclang-cpp14 (1:14.0.6-16) ...
Selecting previously unselected package libopencl-clang14:amd64.
Preparing to unpack .../4-libopencl-clang14_14.0.0-4_amd64.deb ...
Unpacking libopencl-clang14:amd64 (14.0.0-4) ...
Selecting previously unselected package libigdfcl1:amd64.
Preparing to unpack .../5-libigdfcl1_1.0.12504.6-1_amd64.deb ...
Unpacking libigdfcl1:amd64 (1.0.12504.6-1) ...
Selecting previously unselected package intel-opencl-icd.
Preparing to unpack .../6-intel-opencl-icd_22.43.24595.41-1_amd64.deb ...
Unpacking intel-opencl-icd (22.43.24595.41-1) ...
Setting up libllvm14:amd64 (1:14.0.6-16) ...
Setting up libllvmspirvlib14:amd64 (14.0.0-9) ...
Setting up libigc1:amd64 (1.0.12504.6-1) ...
Setting up libclang-cpp14 (1:14.0.6-16) ...
Setting up libopencl-clang14:amd64 (14.0.0-4) ...
Setting up libigdfcl1:amd64 (1.0.12504.6-1) ...
Setting up intel-opencl-icd (22.43.24595.41-1) ...
Processing triggers for libc-bin (2.38-1ubuntu6) ...
/sbin/ldconfig.real: /usr/lib/wsl/lib/libcuda.so.1 is not a symbolic link

Scanning processes...

No services need to be restarted.

No containers need to be restarted.

No user sessions are running outdated binaries.

No VM guests are running outdated hypervisor (qemu) binaries on this host.
edmondium@LAPTOP-1Q9H40K6:~$ clinfo
Abort was called at 54 line in file:
./shared/source/os_interface/windows/wddm/create_um_km_data_translator.cpp
Aborted
edmondium@LAPTOP-1Q9H40K6:~$

JablonskiMateusz, 2023-11-09:
What driver do you have on Windows? 
Could you run `strace -o strace.log clinfo` and share strace.log  file ?

edmondium, 2023-11-09:
> What driver do you have on Windows? Could you run `strace -o strace.log clinfo` and share strace.log file ?

[strace.log](https://github.com/intel/compute-runtime/files/13309424/strace.log)
I have some drivers:
- NVIDIA GeForce Game Ready Driver v546.01
- Intel® Graphics Driver 31.0.101.2115 for 6th-10th Gen Intel® Core™ Processor Graphics

JablonskiMateusz, 2023-11-14:
could you share device id of Intel GPU card you have in the system?

edmondium, 2023-11-14:
> could you share device id of Intel GPU card you have in the system?

how to know device ID?

JablonskiMateusz, 2023-11-15:
@edmondium you may check that in device manager in display adapter's properties:

[image]

BTW. I commited fix for handling missing library not to abort the entire execution: https://github.com/intel/compute-runtime/commit/c454e0d33e0a4c731ba2e2d4eb220c7b1210739e Could you please verify if it helps you get rid of abort?

edmondium, 2023-11-15:
> @edmondium you may check that in device manager in display adapter's properties:
> 
> [image]
> 
> BTW. I commited fix for handling missing library not to abort the entire execution: [c454e0d](https://github.com/intel/compute-runtime/commit/c454e0d33e0a4c731ba2e2d4eb220c7b1210739e) Could you please verify if it helps you get rid of abort?

here is the Intel device ID:
PCI\VEN_8086&DEV_3E9B&SUBSYS_85FC103C&REV_00
PCI\VEN_8086&DEV_3E9B&SUBSYS_85FC103C
PCI\VEN_8086&DEV_3E9B&CC_030000
PCI\VEN_8086&DEV_3E9B&CC_0300

I install the "intel-opencl-icd" from ubuntu repository, precisely from mantic distribution. The "intel-opencl-icd" version is 22.43.24595.41-1: amd64.

JablonskiMateusz, 2024-01-25:
@edmondium could you please install out latest release from github and verify if the issue is gone?

edmondium, 2024-05-08:
Updating the Intel graphics driver solves my problem.
When I updated my ubuntu to latest version (Ubuntu 24.04 LTS), the problem still persisted (the Intel OpenCL ICD was upgrading to version 23.43.27642.40 automatically).
My laptop have 2 graphics driver (Intel and NVIDIA). I suspect the NVIDIA driver will make the problem appear again when I update the driver.
I think the Intel OpenCL ICD should have a mechanism to detect its OpenCL, not NVIDIA's OpenCL.