intel/compute-runtime #692: zeModuleDynamicLink returns ZE_RESULT_ERROR_UNSUPPORTED_FEATURE

linehill, 2023-11-17:
zeModuleDynamicLink returns ZE_RESULT_ERROR_UNSUPPORTED_FEATURE for HD Graphics 530. The error code is not listed in https://spec.oneapi.io/level-zero/latest/core/api.html#zemoduledynamiclink. Is this intended behavior? Should the error code be listed in the function's description?

JablonskiMateusz, 2023-11-17:
Hi @linehill  could you share driver version?

linehill, 2023-11-17:
I'm currently using Intel Compute Runtime 23.26.26690.22. The driver version is 1.3.26690 as reported by zeDriverGetProperties.

JablonskiMateusz, 2023-11-17:
@linehill  
I agree that ZE_RESULT_ERROR_UNSUPPORTED_FEATURE  is not the best return value and it should be changed to ZE_RESULT_ERROR_MODULE_LINK_FAILURE. 

However, looking at our code I see only one path where we return unsupported feature: https://github.com/intel/compute-runtime/blob/master/level_zero/core/source/module/module_imp.cpp#L1425-L1427C16 

could you try to compile the module with "-library-compilation" flag?

linehill, 2023-11-20:
> could you try to compile the module with "-library-compilation" flag?

The flag solved the issue. Thanks!

JablonskiMateusz, 2023-11-21:
fix merged: https://github.com/intel/compute-runtime/commit/914a1f980f423f3bd2b40b912f637be06ca195b3

linehill, 2023-11-23:
Btw, should the `-library-compilation` flag be documented in the Level Zero specification? It doesn't seem to be mentioned there.

JablonskiMateusz, 2023-11-24:
you can find some documentation in our programmers guide: https://github.com/intel/compute-runtime/blob/master/programmers-guide/MODULE_SYMBOL_SUPPORT.md

closed 2024-01-25 by JablonskiMateusz