intel/compute-runtime #697: UHD Graphics 600 - OUT_OF_RESOURCES error.

GiorgosXou, 2024-01-16:
https://github.com/inducer/pyopencl/issues/711#issuecomment-1892805510

JablonskiMateusz, 2024-01-25:
Hi @GiorgosXou  

Could you share more details about the issue?
- repro steps
- OCL API logs captured with clIntercept (https://github.com/intel/opencl-intercept-layer/blob/main/docs/controls.md)

OUT_OF_RESOURCES error returned from clFinish may indicate that there was gpu hang. Could you check if there is any info in dmesg?

GiorgosXou, 2024-01-26:
Hi @JablonskiMateusz 

A way to reproduce the issue is either by running the `cl-demo` from this [repo](https://github.com/hpc12/tools/tree/master) or *([in pyopencl] as I also explain [here](https://github.com/inducer/pyopencl/issues/711#issue-2082578984))* by calling a kernel + enqueue_copy more than once.

***Here's what `dmesg` showed:***
```
[ 4686.449031] i915 0000:00:02.0: [drm] Resetting rcs0 for CS error
[ 4686.449115] i915 0000:00:02.0: [drm] cl-demo[8038] context reset due to GPU hang
[ 4686.453976] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:8ed1fff2, in cl-demo [8038]
[ 4686.453981] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[ 4686.453983] Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/intel/issues/new.
[ 4686.453984] Please see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[ 4686.453985] drm/i915 developers can then reassign to the right component if it's not a kernel issue.
[ 4686.453986] The GPU crash dump is required to analyze GPU hangs, so please always attach it.
[ 4686.453988] GPU crash dump saved to /sys/class/drm/card1/error
```
***GPU crash dump:***
```
GPU HANG: ecode 9:1:8ed1fff2, in cl-demo [8038]
Kernel: 6.7.1-arch1-1 x86_64
Driver: 20230929
Time: 1706289681 s 334166 us
Boottime: 4686 s 550412 us
Uptime: 4679 s 676867 us
Capture: 4296283204 jiffies; 208697 ms ago
Active process (on ring rcs0): cl-demo [8038]
Reset count: 0
Suspend count: 0
Platform: GEMINILAKE
Subplatform: 0x0
PCI ID: 0x3185
PCI Revision: 0x03
PCI Subsystem: 8086:2212
IOMMU enabled?: 0
DMC initialized: yes
DMC loaded: yes
DMC fw version: 1.4
RPM wakelock: yes
PM suspended: no
IER: 0x08080000
DERRMR: 0xffffffff
GT awake: yes
CS timestamp frequency: 0 Hz, 0 ns
EIR: 0x00000000
PGTBL_ER: 0x00000000
GTIER[0]: 0x09090909
GTIER[1]: 0x09090909
GTIER[2]: 0x00000000
GTIER[3]: 0x00000909
  fence[0] = 00000000
  fence[1] = 108603b00880001
  fence[2] = 00000000
  fence[3] = 00000000
  fence[4] = 00000000
  fence[5] = 00000000
  fence[6] = 00000000
  fence[7] = 00000000
  fence[8] = 00000000
  fence[9] = 00000000
  fence[10] = 00000000
  fence[11] = 00000000
  fence[12] = 00000000
  fence[13] = 00000000
  fence[14] = 00000000
  fence[15] = 00000000
  fence[16] = 00000000
  fence[17] = 00000000
  fence[18] = 00000000
  fence[19] = 00000000
  fence[20] = 00000000
[... 162 lines cut ...]
i915.guc_log_level=-1
i915.guc_firmware_path=(null)
i915.huc_firmware_path=(null)
i915.dmc_firmware_path=(null)
i915.gsc_firmware_path=(null)
i915.memtest=no
i915.mmio_debug=0
i915.edp_vswing=0
i915.reset=3
i915.inject_probe_failure=0
i915.enable_dpcd_backlight=-1
i915.force_probe=*
i915.request_timeout_ms=20000
i915.lmem_size=0
i915.lmem_bar_size=0
i915.enable_hangcheck=yes
i915.load_detect_test=no
i915.force_reset_modeset_test=no
i915.error_capture=yes
i915.disable_display=no
i915.verbose_state_checks=yes
i915.nuclear_pageflip=no
i915.enable_dp_mst=yes
i915.enable_gvt=no
```

nyanmisaka, 2024-01-27:
Duplicate of #679

@JablonskiMateusz 
GekminiLake has been failing on the mainline kernel for some time. The 5.15 kernel that comes with Ubuntu 22.04 LTS is not affected, I guess that's why Intel hasn't noticed this issue yet.

> All platforms were validated on Ubuntu 22.04 LTS with stock kernel, unless noted otherwise.

closed 2025-12-03 by kgibala