intel/compute-runtime #702: MTL Number of platforms 0 (GPU not detected?)

soehrlein, 2024-01-31:
Using the latest build (23.48.27912.11) , on kernel 6.8.0-rc2-2024-01-29-intel-next+ , on Ubuntu 22.04.3 LTS, cannot run OpenCL applications.

[   11.097915] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   11.103789] i915 0000:00:02.0: vgaarb: deactivate vga console
[   11.114768] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   11.137904] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[   11.154970] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.17)
[   12.009835] i915 0000:00:02.0: [drm] [ENCODER:235:DDI A/PHY A] failed to retrieve link info, disabling eDP
[   13.873853] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin (70.12.1) is recommended, but only i915/mtl_guc_70.bin (70.8.0) was found
[   13.887004] i915 0000:00:02.0: [drm] GT0: Consider updating your linux-firmware pkg or downloading from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   13.904284] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin (70.12.1) is recommended, but only i915/mtl_guc_70.bin (70.8.0) was found
[   13.917416] i915 0000:00:02.0: [drm] GT1: Consider updating your linux-firmware pkg or downloading from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   13.938212] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.8.0
[   13.957555] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   13.963585] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   13.969408] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[   13.982010] i915 0000:00:02.0: [drm] GT1: can't init GSC proxy due to missing mei component
[   13.990477] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.8.0
[   13.998560] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[   14.032218] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
[   14.040208] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[   14.046232] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[   14.051927] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[   14.069394] [drm] Initialized i915 1.6.0 20230929 for 0000:00:02.0 on minor 0
[   14.095340] i915 display info: display version: 14
[   14.095343] i915 display info: cursor_needs_physical: no
[   14.095344] i915 display info: has_cdclk_crawl: yes
[   14.095345] i915 display info: has_cdclk_squash: yes
[   14.095346] i915 display info: has_ddi: yes
[   14.095346] i915 display info: has_dp_mst: yes
[   14.095347] i915 display info: has_dsb: yes
[   14.095348] i915 display info: has_fpga_dbg: yes
[   14.095349] i915 display info: has_gmch: no
[   14.095350] i915 display info: has_hotplug: yes
[   14.095351] i915 display info: has_hti: no
[   14.095352] i915 display info: has_ipc: yes
[   14.095353] i915 display info: has_overlay: no
[   14.095353] i915 display info: has_psr: yes
[   14.095354] i915 display info: has_psr_hw_tracking: no
[   14.095354] i915 display info: overlay_needs_physical: no
[   14.095355] i915 display info: supports_tv: no
[   14.095355] i915 display info: has_hdcp: yes
[   14.095356] i915 display info: has_dmc: yes
[   14.095356] i915 display info: has_dsc: yes
[   14.202889] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.0.0.1655, svn 1)
[   14.308513] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[   14.315662] i915 0000:00:02.0: [drm] *ERROR* GT1: GSC proxy handler failed to init
[   14.973631] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   15.880447] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   16.748409] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   19.736764] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

With clinfo:

execve("/usr/bin/clinfo", ["clinfo"], 0x7fff638b1c20 /* 18 vars */) = 0
brk(NULL)                               = 0x55b443b42000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffe005bd560) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9de0bc3000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=24287, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 24287, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f9de0bbd000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libOpenCL.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300c\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=69288, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 71200, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f9de0bab000
mmap(0x7f9de0bb1000, 24576, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7f9de0bb1000
mmap(0x7f9de0bb7000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xc000) = 0x7f9de0bb7000
mmap(0x7f9de0bbb000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xf000) = 0x7f9de0bbb000
[... 545 lines cut ...]
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7fffa27fdce0) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7fffa27fdd00) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fffa27fdc90) = 0
futex(0x55bc61efee10, FUTEX_WAKE_PRIVATE, 2147483647) = 0
sysinfo({uptime=1392, loads=[0, 0, 0], totalram=67103035392, freeram=66101899264, sharedram=24244224, bufferram=35807232, totalswap=8589930496, freeswap=8589930496, procs=337, totalhigh=0, freehigh=0, mem_unit=1}) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7fffa27fe540) = 0
ioctl(4, DRM_IOCTL_I915_GEM_VM_DESTROY, 0x7fffa27fe650) = 0
close(4)                                = 0
close(3)                                = 0
futex(0x7efef116f210, FUTEX_WAKE_PRIVATE, 2147483647) = 0
write(2, "terminate called after throwing "..., 48terminate called after throwing an instance of ') = 48
write(2, "cl::Error", 9cl::Error)                = 9
write(2, "'\n", 2'
)                      = 2
write(2, "  what():  ", 11  what():  )             = 11
write(2, "clGetPlatformIDs", 16clGetPlatformIDs)        = 16
write(2, "\n", 1
)                       = 1
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
gettid()                                = 1302
getpid()                                = 1302
tgkill(1302, 1302, SIGABRT)             = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=1302, si_uid=0} ---
+++ killed by SIGABRT (core dumped) +++
Aborted (core dumped)

soehrlein, 2024-02-01:
Installing Ubuntu 23.10 (kernel 6.5.0-9), adding i915.force_probe=7d55, and then installing manually all of the *.deb files from the compute-runtime seems to work. I'm not sure why, and will at some point need to recompile a newer kernel, hopefully it continues to work with a 6.7/6.8-intel-next kernel.

Leaving this open momentarily to see if anyone might suggest what could have been wrong, then will close.

thetick1234567890, 2024-02-07:
Hi,

I'm also am seeing no devices found with 6.8rc1 / rc 2 / rc3 Ubuntu mainline kernels.  The exact same hardware rebooted with 6.7.3 works just fine with opencl workloads.  I tried on two different old but supported iGPU ( Broadwell, Skylake) platforms.

Could you please try and run clinfo on a 6.8 kernel ?  I believe the issue with no found devices should be easy to reproduce.

soehrlein, 2024-02-07:
Sorry it was difficult to find but if you search clinfo on my original post it contained the clinfo output from 6.8.

thetick1234567890, 2024-02-13:
Hi,

@soehrlein I'm just a user like you.  I was just adding that I also was having issues with 6.8rc kernels.  My request above was for the Intel developers.

closed 2024-02-13 by soehrlein