intel/compute-runtime #703: FAQ says implementation pending for cl_cache for Windows

apartridge, 2024-02-01:
In the FAQ it states that Windows implementation of cl_cache is pending: https://github.com/intel/compute-runtime/blob/master/opencl/doc/FAQ.md#windows

It seems that since https://github.com/intel/compute-runtime/commit/d7d577ba4873b83cf57367529b49384c9748d510 the cl_cache feature is also enabled by default on Windows. I tested this feature (on both Linux and Windows) and it seems to be working fine.

So I think the FAQ document is just a bit out of date. and can be updated?

JablonskiMateusz, 2024-02-02:
Hi @apartridge
Thank you for catching that!
The feature is enabled for OCL by default, however L0 part is still in progress. We will update documentation once the feature is fully implemented.

JablonskiMateusz, 2024-04-24:
Documentation updated: https://github.com/intel/compute-runtime/commit/9553dd1098ba624090541860d2de75eac2d1ee71

closed 2024-04-25 by JablonskiMateusz