intel/compute-runtime #704: build error error: undefined reference to `_Z18__spirv_ImageWritePU3AS133__spirv_Image__void_2_0_0_0_0_0_1Dv4_iDv4_ji'

pvelesko, 2024-02-02:
[neo-error.txt](https://github.com/intel/compute-runtime/files/14142301/neo-error.txt)

Tried with building LLVM standalone and also with LLVM getting built alongside neo.

JablonskiMateusz, 2024-02-05:
Hi @pvelesko 

Could you try with IGC as of https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.15770.11  ?
In release notes we provide which version of IGC was used for given release

dbermond, 2024-02-15:
Not sure if this is related, but I'm getting very similar errors in Arch Linux since version [23.52.28202.14](https://github.com/intel/compute-runtime/releases/tag/23.52.28202.14).

Version [24.05.28454.6](https://github.com/intel/compute-runtime/releases/tag/24.05.28454.6) + IGC [1.0.15985.7](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.15985.7) also gives me the same result in when building:

```
error: undefined reference to `__builtin_spirv_OpenCL_vload4_i64_p1i32_i32'
in function: '__builtin_spirv_OpenCL_vload4_i64_p1i32_i32' called by kernel: 'fullCopy'

error: backend compiler failed build.

error: undefined reference to `__builtin_spirv_OpenCL_vstore4_v4i32_i64_p1i32'
in function: '__builtin_spirv_OpenCL_vstore4_v4i32_i64_p1i32' called by kernel: 'fullCopy'

error: backend compiler failed build.

error: undefined reference to `__builtin_spirv_BuiltInGlobalInvocationId'
in function: '__builtin_spirv_BuiltInGlobalInvocationId' called by kernel: 'fullCopy'

error: backend compiler failed build.

Build failed with error code: -11
Command was: /build/intel-compute-runtime/src/build/bin/ocloc -q -file /build/intel-compute-runtime/src/build/bin/built_ins/x64/xe_hpc_core/stateless_aux_translation_pvc.spv -spirv_input -device 12.60.0 -cl-intel-greater-than-4GB-buffer-required -64 -output stateless_aux_translation_12_60_0 -output_no_suffix -out_dir /build/intel-compute-runtime/src/build/bin/built_ins/x64/xe_hpc_core -options "-cl-kernel-arg-info"
```

There are other similar errors following it in the compilation, like these:

```
error: undefined reference to `__builtin_spirv_BuiltInGlobalInvocationId'
in function: '__builtin_spirv_BuiltInGlobalInvocationId' called by kernels:
  - 'CopyBufferRectBytes2d'
  - 'CopyBufferRectBytes3d'
```

dbermond, 2024-03-29:
@JablonskiMateusz Any update on this?

I tried using latest neo 24.09.28717.12 with igc 1.0.16238.4, and the problem still persists. Both neo and igc are built with the listed dependency commits from their release description.

Yanfeng-Mi, 2024-04-01:
I got the same issue too, any solutions for this issue?

vidalinux, 2024-04-04:
Same issue here in Arch Linux with Version 24.09.28717.12 + ICG 1.0.16510.1

vidalinux, 2024-04-08:
@dbermond  I was able to fix the compile issues on Arch Linux updating all PKGBUILD for IGC + dependencies, changes are available here: https://github.com/vidalinux/AUR:

JablonskiMateusz, 2024-04-09:
> @JablonskiMateusz Any update on this?

The error message says directly that this is failure on compiler backend side. 
In the log there is also cmdline to reproduce the issue
> Command was: /home/pvelesko/space/intel-compute-runtime-build/neo/build/bin/ocloc -q -file /home/pvelesko/space/intel-compute-runtime-build/neo/build/bin/built_ins/x64/gen8/stateless_copy_buffer_to_image3d_stateless_bdw.spv -spirv_input -device 8.0.0 -cl-intel-greater-than-4GB-buffer-required -64 -output stateless_copy_buffer_to_image3d_stateless_8_0_0 -output_no_suffix -out_dir /home/pvelesko/space/intel-compute-runtime-build/neo/build/bin/built_ins/x64/gen8 -options "-cl-kernel-arg-info"

Please attach reproducer and create an issue in IGC repo

dbermond, 2024-04-10:
@vidalinux I have not compiled/tested your setup, but currently it's not suitable because you are using different dependency versions for IGC in comparison to the ones specified by upstream. Although this may compile, it can have silent segmentation fault issues that are shown only at runtime, since IGC seems to be sensible to these versions. This have already happened in the past when we used binary dependencies. Also, your spirv-tools is 2024.1, which is ahead of the one shipped in the official repositories. 

We have switched to using a source-based build for most IGC dependencies exactly for, among other things, avoid the problems described above, making sure that we are using dependencies that are tested by upstream, and the current intention is to stick with this type of build. This also helped a lot with building issues in IGC when we were using a binary (prebuild) llvm at a different version. For using your setup, we would need a spirv-llvm-translator15 (and so on) in the official repositories, and we are avoiding this by using a source-based build.

pvelesko, 2024-04-10:
@dbermond  @JablonskiMateusz  an issue has been open on igc for a while.

dbermond, 2024-04-28:
It works fine for me when using `IGC_OPTION__USE_KHRONOS_SPIRV_TRANSLATOR_IN_SC=OFF` in IGC.

closed 2025-12-03 by kgibala