intel/compute-runtime #706: Runtime hangs on DG2 (and Gen12 iGPU maybe?)

tazz4843, 2024-02-09:
I'm running into random hangs when my app is running during normal use, that began occurring several months ago, roughly September 2023. A stack trace is attached, see end for it. I was doing some digging and found [this related comment](
https://github.com/intel/compute-runtime/issues/559#issuecomment-1271027342) with the exact same stack trace, although only on DG2 and running an unsupported kernel, while I was able to occasionally reproduce this on Gen12 iGPUs and on a much more modern kernel version.
I'm using [`whisper.cpp`](https://github.com/ggerganov/whisper.cpp) with its OpenCL backend to run arbitrary speech-to-text. If one thread ends up hanging, all other runtime threads also end up hanging, spinning multiple cores to 100%.

I'm very new to all of this so please let me know if there's any information I can supply :)

Host details:
GPU: Arc A770
Arch Linux w/ kernel 6.7.3-arch1-1.1
intel-compute-runtime-23.48.27912.11-1

[backtrace.txt](https://github.com/intel/compute-runtime/files/14223856/scratch_1.txt)

eero-t, 2024-02-13:
Looking at the backtrace:
* 8 "tokio-runtime-w" threads have yielded their execution in `NEO::CommandStreamReceiver::baseWaitFunction()`
* 1 "scripty_stt_ser" thread is futex waiting worker closing in `NEO::DrmGemCloseWorker::worker()`
* 1 "scripty_stt_ser" thread is Tokyo Rust code itself directly hanging in `futex_wait()` syscall

geekboood, 2024-02-21:
I have the same issue when running openvino model server

tazz4843, 2024-02-21:
Sorry this took me so long to get back to.

> Looking at the backtrace:
> 
> * 1 "scripty_stt_ser" thread is Tokyo Rust code directly hanging in `futex_wait()` syscall

From what I've looked at the code, it seems that this runtime worker is waiting for compute runtime code to return thus making me think this is the issue. Disabling the OpenCL runtime and falling back to CPU makes this issue completely disappear, even after weeks of runtime, compared to usually at most 1 week before it locks up and starts spinning on CPU with OpenCL integration.

tazz4843, 2025-10-29:
It's been 1.5 years, I've since switched to using an AMD GPU to avoid this bug and no longer have access to an Arc GPU.

kgibala, 2025-12-03:
Hi @tazz4843 ,

Thank you for confirmation, we are closing this issue as we are not able to reproduce it locally. 

If you encounter this or a similar problem in the future, please feel free to create a new issue. When submitting, please follow our [Issue Submission Guide](https://github.com/intel/compute-runtime/blob/master/ISSUE_SUBMISSION_GUIDE.md) to help us understand and address your concern efficiently.