intel/compute-runtime #709: split libze-intel-gpu as a separate source

tjaalton, 2024-02-26:
Hi,

I was told that the Debian/Ubuntu packaging of compute-runtime should jump through hoops and version libze-intel-gpu* with a version number of it's own, which is kinda silly since that'd have to be done by every downstream.

Instead, you should split it out from compute-runtime.

eero-t, 2024-02-28:
Related: https://github.com/intel/compute-runtime/issues/529

eero-t, 2024-04-25:
> I was told that the Debian/Ubuntu packaging of compute-runtime should jump through hoops and version libze-intel-gpu* with a version number of it's own, which is kinda silly since that'd have to be done by every downstream.

Can you give link to that discussion?  Users need to know from which version of `compute-runtime` both of the produced drivers come from, _especially_ L0 one, and if they come from same build I think they should use the same version.  Don't see any reason why one would even want to do otherwise...

tjaalton, 2024-04-25:
It was info I got on a private chat with a colleague who was in contact with Intel...