intel/compute-runtime #716: Comet Lake (i7-8850H, UHD 630) support on Arch Linux?

techflashYT, 2024-03-27:
No matter what I do, for the life of me I can't get this to work with my UHD 630 graphics, `clinfo` always returns 0 platforms found.
I've tried the suggestions in other issues here of running `clinfo` as root, and adding my user to the `render` or `video` groups, but yet nothing works.
Any ideas?

the log.txt from `strace -f -s 255 -o log.txt clinfo` is attached below.
[log.txt](https://github.com/intel/compute-runtime/files/14769222/log.txt)

techflashYT, 2024-03-27:
(Fixed title, wrote this at 11:48 PM and mixed up the two somehow...)

smunaut, 2024-03-28:
Did you see #710  ?
Maybe you're affected depending on your kernel version.

techflashYT, 2024-04-15:
Possible, I'll check on it when I have a chance, I looked around didn't spot that as being related.  I'll try the various fixes listed in there when I have time.

eero-t, 2024-04-16:
> Possible, I'll check on it when I have a chance, I looked around didn't spot that as being related. I'll try the various fixes listed in there when I have time.

If you're using 6.8 or newer kernel, you're affected by the bug.

It can be trivially checked with this temporary workaround until fix[1] is available:
```
export NEOReadDebugKeys=1
export OverrideGpuAddressSpace=48
```

[1] There are tagged versions with a fix, but not yet a validated release with driver binary packages.

JablonskiMateusz, 2024-04-24:
@techflashYT  do you see the issue with most recent release?

techflashYT, 2024-04-29:
I can confirm that OpenCL does in fact work on this hardware as of the latest update, or at least, it seems to.  I get `CL_OUT_OF_RESOURCE` errors, but that's hardly unexpected with my hardware.  It seems fine enough.  At least `clinfo` runs, which is an improvement from before.  I presume this issue is now closed then, as the original bug has been fixed.

closed 2024-04-29 by techflashYT