intel/compute-runtime #718: /usr/bin/ld.gold: error: cannot find -ligdgmm

cmsanta, 2024-04-03:
csanta@galathea:~/linux-graphics/opencl/workspace/neo/build$ make
[  4%] Built target ocloc_lib
[  4%] Built target ocloc
[ 16%] Built target neo_shared
[ 16%] Built target cpp_generate_tool
[ 16%] Built target copy_compiler_files
[ 18%] Built target builtins_spirv
[ 18%] Built target igdrcl_lib_release_sharings_enable
[ 21%] Built target builtins_pvc_stateless
[ 21%] Built target builtins_pvc
[ 22%] Built target builtins_bdw_bindful
[ 22%] Built target builtins_bdw_stateless
[ 22%] Built target builtins_bdw
[ 23%] Built target builtins_skl_bindless
[ 23%] Built target builtins_skl_stateless
[ 25%] Built target builtins_skl_bindful
[ 25%] Built target builtins_skl
[ 26%] Built target builtins_kbl_bindless
[ 27%] Built target builtins_kbl_stateless
[ 27%] Built target builtins_kbl_bindful
[ 27%] Built target builtins_kbl
[ 28%] Built target builtins_glk_bindless
[ 29%] Built target builtins_glk_stateless
[ 29%] Built target builtins_glk_bindful
[ 29%] Built target builtins_glk
[ 30%] Built target builtins_cfl_bindless
[ 30%] Built target builtins_cfl_stateless
[ 31%] Built target builtins_cfl_bindful
[ 31%] Built target builtins_cfl
[ 31%] Built target builtins_bxt_bindless
[ 32%] Built target builtins_bxt_stateless
[ 33%] Built target builtins_bxt_bindful
[ 33%] Built target builtins_bxt
[ 34%] Built target builtins_icllp_bindless
[ 35%] Built target builtins_icllp_stateless
[ 35%] Built target builtins_icllp_bindful
[ 35%] Built target builtins_icllp
[ 35%] Built target builtins_lkf_bindless
[ 36%] Built target builtins_lkf_stateless
[ 38%] Built target builtins_lkf_bindful
[ 38%] Built target builtins_lkf
[ 39%] Built target builtins_ehl_bindless
[ 40%] Built target builtins_ehl_stateless
[ 41%] Built target builtins_ehl_bindful
[ 41%] Built target builtins_ehl
[ 42%] Built target builtins_tgllp_bindless
[ 42%] Built target builtins_tgllp_stateless
[ 43%] Built target builtins_tgllp_bindful
[ 43%] Built target builtins_tgllp
[ 44%] Built target builtins_dg1_bindless
[ 44%] Built target builtins_dg1_stateless
[ 45%] Built target builtins_dg1_bindful
[ 45%] Built target builtins_dg1
[ 46%] Built target builtins_rkl_bindless
[ 46%] Built target builtins_rkl_stateless
[ 47%] Built target builtins_rkl_bindful
[ 47%] Built target builtins_rkl
[ 48%] Built target builtins_adls_bindless
[ 48%] Built target builtins_adls_stateless
[ 50%] Built target builtins_adls_bindful
[ 50%] Built target builtins_adls
[ 51%] Built target builtins_adlp_bindless
[ 51%] Built target builtins_adlp_stateless
[ 52%] Built target builtins_adlp_bindful
[ 52%] Built target builtins_adlp
[ 53%] Built target builtins_adln_bindless
[ 54%] Built target builtins_adln_stateless
[ 54%] Built target builtins_adln_bindful
[ 54%] Built target builtins_adln
[ 56%] Built target builtins_mtl_bindless
[ 57%] Built target builtins_mtl_stateless
[ 60%] Built target builtins_mtl_bindful
[ 60%] Built target builtins_mtl
[ 66%] Built target builtins_dg2_bindless
[ 68%] Built target builtins_dg2_stateless
[ 73%] Built target builtins_dg2_bindful
[ 73%] Built target builtins_dg2
[ 75%] Built target builtins_arl_bindless
[ 76%] Built target builtins_arl_stateless
[ 78%] Built target builtins_arl_bindful
[ 78%] Built target builtins_arl
[ 78%] Built target builtins
[ 83%] Built target builtins_binaries_stateless
[ 83%] Built target builtins_vme_sources
[ 93%] Built target builtins_binaries_bindful
[ 93%] Built target builtins_sources
[ 96%] Built target igdrcl_lib_release
[ 96%] Built target builtins_vme
[ 96%] Linking CXX shared library ../bin/libigdrcl.so
/usr/bin/ld.gold: error: cannot find -ligdgmm
collect2: error: ld returned 1 exit status
make[2]: *** [igdrcl_lib_release/CMakeFiles/igdrcl_dll.dir/build.make:3124: bin/libigdrcl.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:4945: igdrcl_lib_release/CMakeFiles/igdrcl_dll.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
csanta@galathea:~/linux-graphics/opencl/workspace/neo/build$ ls /usr/local/lib/
igc            libiga64.so.1.0.1  libigc.so.1.0.1  libigdfcl.so.1.0.1  libigdgmm.so.12.3.0    python3.10
libiga64.so    libigc.so          libigdfcl.so     libigdgmm.so        libopencl-clang.so.14  x86_64-linux-gnu
libiga64.so.1  libigc.so.1        libigdfcl.so.1   libigdgmm.so.12     pkgconfig
csanta@galathea:~/linux-graphics/opencl/workspace/neo/build$

cmsanta, 2024-04-03:
the build system can't seem to find igdgmm library even though it's under /usr/local/lib directory

Does the build need to know about this library with pkconfig? if so, how can I do that?

cmsanta, 2024-04-03:
[CMakeCache.txt](https://github.com/intel/compute-runtime/files/14858999/CMakeCache.txt)

JablonskiMateusz, 2024-04-09:
hi @cmsanta 
please try refreshing dynamic linker `sudo ldconfig` 
if it doesn't help then please get cmdline from linking libigdrcl.so and verify if path is correct

chao-camect, 2024-04-20:
ldconfig doesn't help. This is new in 24.13.29138.7. Was fine in the previous release.

eero-t, 2024-04-22:
> ldconfig doesn't help. This is new in 24.13.29138.7. Was fine in the previous release.

Does `ldconfig -v` list the library?

And if not, is that path in your ldconfig configuration?
`head /etc/ld.so.conf /etc/ld.so.conf.d/*`

chao-camect, 2024-07-06:
I have the same issue. It's the same for 24.22.29735.20.
the so is in `ldconfig -v`

mikamiel, 2024-09-01:
The problem is that in Ubuntu 24.04 (at least) `ld.gold` linker is not informed about search path `/usr/local/lib` (gmmlib library build script installs `libigdgmm.so` in that dir by default). To workaround - just copy or link file `libigdgmm.so` into `/usr/lib` folder.

GermanAizek, 2024-12-18:
Same error on Debian Testing (sid) 6.12.5-1 (2024-12-15) x86_64 GNU/Linux

eero-t, 2024-12-18:
> The problem is that in Ubuntu 24.04 (at least) `ld.gold` linker is not informed about search path `/usr/local/lib` (gmmlib library build script installs `libigdgmm.so` in that dir by default). I didn't find a way to add search path to `ld.gold` linker in Ubuntu globally or through env variables.

That Gold linker bug is at least 13 years old: https://stackoverflow.com/questions/7435548/gold-linker-on-ubuntu-11-04-does-not-search-usr-local-lib-by-default

One would have expected it have been fixed already by now...

=> @mikamiel & @GermanAizek What's the output of: `head /etc/ld.so.conf /etc/ld.so.conf.d/*` ?

@tjaalton Any suggestions?

pwilma, 2025-10-29:
Hi @cmsanta ,
We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!