intel/compute-runtime #720: How to verify GPU driver is properly installed on Meteor Lake Platform

avitial, 2024-04-15:
Trying to run inference with OpenVINO on GPU but unable to do so as we were told the GPU driver does not seem to be properly installed on the platform. 

We have followed instructions as per [configurations-intel-gpu](https://docs.openvino.ai/2024/get-started/configurations/configurations-intel-gpu.html) and OpenVINO is unable to recognize the GPU device. We have tried different compute-runtime versions (most recently v1.0.16238.4) and unable to figure out why system cannot detect GPU. Even built OpenVINO 2024.0 from source and the result is the same, we are unable to execute inference using iGPU. 

How can verify the GPU drivers are properly installed on the system? Is there a comprehensive list of tasks to get MTL GPU driver setup? Thanks in advance.

```
$ clinfo
Number of platforms                               0

$ lscpu
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          46 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   22
  On-line CPU(s) list:    0-21
Vendor ID:                GenuineIntel
  Model name:             Intel(R) Core(TM) Ultra 7 155H
    CPU family:           6
    Model:                170
    Thread(s) per core:   2
    Core(s) per socket:   16
    Socket(s):            1
    Stepping:             4
[...]

$ uname -a
Linux aisw-msi-ultra7-155h-d 6.8.6-zabbly+ #ubuntu22.04 SMP PREEMPT_DYNAMIC Sun Apr 14 02:03:36 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/*release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"
PRETTY_NAME="Ubuntu 22.04.4 LTS"
[...]
```

eero-t, 2024-04-16:
> Linux aisw-msi-ultra7-155h-d 6.8.6-zabbly+

You're hitting bug #710.  Workaround until there's a fixed driver release [1]:
```
export NEOReadDebugKeys=1
export OverrideGpuAddressSpace=48
```

[1] If you can build your own drivers, 24.09.28717.14 tag or later ones, include a fix: https://github.com/intel/compute-runtime/tags

avitial, 2024-04-19:
@JablonskiMateusz yes it did, I built with tag 24.09.28717.14 and applied the workaround provided by @eero-t and now OpenVINO can detect iGPU device. When is this fix making it to master branch? Thank you so much.

[image]

avitial, 2024-04-24:
@eero-t thanks for confirming. One more thing, I've tried this on Ubuntu 22.04 with Kernel 6.8.4 and it is working, but for another system with _Kernel 6.5.0-17-generic_ it is not. The https://github.com/intel/compute-runtime/releases page mentions "_Meteor Lake was tested with [kernel v6.6.7](https://kernel.ubuntu.com/mainline/v6.6.7/)_", is the kernel version >=6.6.7 a requirement for the driver to work? Or should this also work for kernel versions below the tested version? Thanks beforehand for your response!

eero-t, 2024-04-24:
According to Phoronix, stable Meteor Lake support came in 6.7 kernel, with some additional fixes in 6.8:
* https://www.phoronix.com/news/Linux-6.7-Intel-Meteor-Lake-Gfx
* https://www.phoronix.com/news/Linux-6.8-Meteor-Lake-Clocks

So I assume you'll need kernel `force_probe` option for 6.6 kernel (to use the earlier unstable version of the kernel driver).

JablonskiMateusz, 2024-04-24:
@avitial  can we close the issue now?

avitial, 2024-04-26:
I will close the issue for now. We are still working on that other MTL platform with kernel 6.5, so I will reopen if needed. Many thanks for the help!

closed 2024-04-26 by avitial