intel/compute-runtime #721: WSL: zesInit segfaults

tkatila, 2024-04-16:
I'd like to use Sysman API in a container in WSL. I noticed that when I call `zesInit()` the app segmentation faults.

I used the latest 22.04 container + latest compute-runtime release (https://github.com/intel/compute-runtime/releases/tag/24.09.28717.12).

Container is started like so: `docker run -v /usr/lib/wsl:/usr/lib/wsl:ro --device /dev/dxg --rm -it ubuntu:22.04 bash`

My app is essentially just a call to zesInit:
```
#include <zes_api.h>
#include <stdio.h>

int main()
{
        ze_result_t res = zesInit(0);
        if (res == 0) {
                fprintf(stderr, "all good\n");
        } else {
                fprintf(stderr, "all bad\n");
        }

        return 0;
}
```

Compiled with `gcc l0.cpp -o l0 -I/usr/include/level_zero -lze_loader`

GDB backtrace is like so:
```
 Program received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
74      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
(gdb) bt
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74
#1  0x00007f673a926789 in std::char_traits<char>::length (__s=0x100000000 <error: Cannot access memory at address 0x100000000>)
    at /usr/include/c++/11/bits/char_traits.h:399
#2  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::allocator<char> > (__a=...,
    __s=0x100000000 <error: Cannot access memory at address 0x100000000>, this=0x56233579dba0) at /usr/include/c++/11/bits/basic_string.h:536
#3  NEO::HwDeviceIdDrm::HwDeviceIdDrm (devNodePathIn=0x0, pciPathIn=0x100000000 <error: Cannot access memory at address 0x100000000>, fileDescriptorIn=<optimized out>,
    this=0x56233579db90) at ../../neo/shared/source/os_interface/linux/hw_device_id.h:24
#4  L0::Sysman::SysmanHwDeviceIdDrm::HwDeviceIdDrm (this=0x56233579db90) at ../../neo/level_zero/sysman/source/shared/linux/sysman_hw_device_id_linux.h:17
#5  std::make_unique<L0::Sysman::SysmanHwDeviceIdDrm, int, char const*, char const*> () at /usr/include/c++/11/bits/unique_ptr.h:962
#6  L0::Sysman::createSysmanHwDeviceId (hwDeviceId=std::unique_ptr<NEO::HwDeviceId> = {...})
    at ../../neo/level_zero/sysman/source/shared/linux/sysman_hw_device_id_linux.cpp:48
#7  0x00007f673a91e0b6 in L0::Sysman::SysmanDriverImp::initialize (this=<optimized out>, result=0x7ffc20c4c504)
    at ../../neo/level_zero/sysman/source/driver/sysman_driver.cpp:49
#8  0x00007f673a91de8f in operator() (__closure=<optimized out>) at ../../neo/level_zero/sysman/source/driver/sysman_driver.cpp:78
#9  std::__invoke_impl<void, L0::Sysman::SysmanDriverImp::driverInit(zes_init_flags_t)::<lambda()> > (__f=...) at /usr/include/c++/11/bits/invoke.h:61
#10 std::__invoke<L0::Sysman::SysmanDriverImp::driverInit(zes_init_flags_t)::<lambda()> > (__fn=...) at /usr/include/c++/11/bits/invoke.h:96
#11 operator() (__closure=<optimized out>) at /usr/include/c++/11/mutex:776
#12 operator() (__closure=0x0) at /usr/include/c++/11/mutex:712
#13 _FUN () at /usr/include/c++/11/mutex:712
#14 0x00007f673c862ee8 in __pthread_once_slow (once_control=0x7f673c3f6aa8 <L0::Sysman::driverImp+8>, init_routine=0x7f673c677d50 <__once_proxy>)
    at ./nptl/pthread_once.c:116
#15 0x00007f673a91e589 in __gthread_once (__func=<optimized out>, __once=0x7f673c3f6aa8 <L0::Sysman::driverImp+8>)
    at /usr/include/x86_64-linux-gnu/c++/11/bits/gthr-default.h:700
#16 std::call_once<L0::Sysman::SysmanDriverImp::driverInit(zes_init_flags_t)::<lambda()> > (__f=..., __once=...) at /usr/include/c++/11/mutex:783
#17 L0::Sysman::SysmanDriverImp::driverInit (flags=<optimized out>, this=0x7f673c3f6aa0 <L0::Sysman::driverImp>)
    at ../../neo/level_zero/sysman/source/driver/sysman_driver.cpp:76
#18 L0::Sysman::init (flags=<optimized out>) at ../../neo/level_zero/sysman/source/driver/sysman_driver.cpp:119
#19 0x00007f673ca27b5b in zesInit () from /lib/x86_64-linux-gnu/libze_loader.so.1
#20 0x0000562332e7617f in main () at l0.cpp:6
```

saik-intel, 2024-04-24:
We will look into this issue , ideally it shouldn't segfault

eero-t, 2024-08-19:
> We will look into this issue , ideally it shouldn't segfault

@saik-intel / @JablonskiMateusz (nearly) 4 months gone, any news?

saik-intel, 2024-08-26:
we didn't fixed yet, will work on lower priority.

JablonskiMateusz, 2025-04-09:
fix merged in https://github.com/intel/compute-runtime/commit/4a67d1c0d55897589797fc30130052c8d0f8e12c 
it will be included in next release (25.13)

pwilma, 2025-10-29:
Hi @tkatila ,
We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!