intel/compute-runtime #727: Enable Xe KMD support by default

eero-t, 2024-04-26:
The other Intel user-space GPU drivers already enable Xe KMD support by default in their latest releases:
* https://github.com/intel/media-driver/pull/1784
* https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/20418

Maybe OpenCL / L0 / Sysman driver should also do that?

JablonskiMateusz, 2024-05-15:
Hi @eero-t 
Xe KMD support is enabled by default since https://github.com/intel/compute-runtime/commit/a2742492ab0cc84dd5a6512caaca4858c92caab8 
It is included in 24.13 release

eero-t, 2024-05-15:
Verified with the latest tag (`24.17.29377.8`) using `drm-tip` kernel HEAD => Xe KMD is supported without it being explicitly enabled with a CMake option.

closed 2024-05-15 by eero-t