intel/compute-runtime #729: OpenCL JIT compiler hangs

tangjinchuan, 2024-04-27:
Dear Intel OpenCL team,
I have a kernel file which can be compiled with AMD 7800XT GPU using less than 100MB memory during compilation using AMD driver. However, by using the Intel runtime on Xe graphics (the latest version), it hangs during compiling and the memory usage is huge. 
Furthermore, it seems that a "-cl-opt-disable" flag can guarantee the kernel is compiled in a very short time. 

I reported the issue previously to https://community.intel.com/t5/GPU-Compute-Software/OpenCL-JIT-compiler-freezing-with-a-kernel/td-p/1589484 . But no response for quite a while.

The freezing kernel is mainly due to BNField12.cl.
[clTest (2).zip](https://github.com/intel/compute-runtime/files/15136846/clTest.2.zip)

Thank you very much!

Best wishes,

Jinchuan

JablonskiMateusz, 2024-05-15:
Hi @tangjinchuan 
Could you please capture callstack of the hang moment?

eero-t, 2024-05-15:
This can be done e.g. by attaching debugger to the process: `gdb /path/to/program/binary $(pidof <binary name>)`

And telling it to provide backtrace (with `bt` command).