intel/compute-runtime #732: spir-v translator drops a required extension somewhere along the way (breaks opencl)

ohsix, 2024-05-15:
don't know if this is the proper place to ask, but

$ clpeak
Platform: Intel(R) OpenCL Graphics
  Device: Intel(R) Iris(R) Xe Graphics
    Driver version  : 24.09.28717.17 (Linux x64)
    Compute units   : 96
    Clock frequency : 1400 MHz
RequiresExtension: Feature requires the following SPIR-V extension:
 SPV_KHR_non_semantic_info

poked around just about every place else, https://github.com/KhronosGroup/SPIRV-LLVM-Translator/pull/2532 is maybe an indication that the extension is now (1.6) assumed in the frontend but not enabled in the backend?

intel compute-runtime & opencl stuff is as driver version says, on fedora 40 (worked on 39, everything is basically a no change rebuild for 40)

please advise, and thank you

eero-t, 2024-05-15:
Tried with following Ubuntu packages:
- clpeak: 1.1.0
- libllvmspirvlib-14-dev: 14.0.0-12build1
- libopencl-clang-14-dev: 14.0.0-4build2

Using my own compute stack build of the latest release tags:
- GMMlib:            intel-gmmlib-22.3.19
- SPIRV-Headers/-Tools:         vulkan-sdk-1.3.283.0
- VC-intrinsics:     v0.18.0
- Graphics Compiler: igc-1.0.16747.7
- Level-Zero API:    v1.17.0
- compute-runtime:   24.17.29377.8

On top of latest drm-tip Xe KMD on TGL-H iGPU => `clpeak` works just fine...

It worked fine also on top of month older compute stack:
- SPIRV-Headers/-Tools:         vulkan-sdk-1.3.280.0
 - Graphics Compiler: igc-1.0.16695.5
 - compute-runtime:   24.13.29138.7

And on top of several months older compute stack + default i915 KMD:
- GMMlib:            intel-gmmlib-22.3.17
- SPIRV-Headers/-Tools:         vulkan-sdk-1.3.275.0
- VC-intrinsics:     v0.16.0
- Graphics Compiler: igc-1.0.16161.0
- Level-Zero API:    v1.15.8
- compute-runtime:   23.52.28202.14

=> If you're using distro packages, please file bug against your distro.

ohsix, 2024-05-15:
that's a whole lot of noise, even asked questions answered in the original post. specifically asking for advice from people that know the code, not anecdotes

eero-t, 2024-05-16:
Note: I'm not related to this project (or other Intel driver projects), just another user of those drivers trying to help you.

You stated that you have compute stack components that do not work together, but did not state which versions those are (except for `compute-runtime` being an older release).

And I tested that your use-case (`clpeak`) works fine with latest versions of everything, and few older combos.

What other advise you expect based on the information you provided?

PS. Distros mix compute stack components differently from what's tested by Intel, but I do not think them to really test resulting stack _that_ much besides making sure builds unit-tests pass... => if your distro-provided compute stack does not work, you should file bug against your distro, not here.

Additional upstream project ticket (like the one here) normally makes sense only after issue is reproduced with the latest release the project offers.   In this case, either with up to date packages from Intel driver package repo [1], or from latest project release [2].

[1] https://dgpu-docs.intel.com/driver/installation.html
[2] https://github.com/intel/compute-runtime/releases

ohsix, 2024-05-17:
forget it

closed 2024-05-17 by ohsix

eero-t, 2024-05-20:
> You stated that you have compute stack components that do not work together, but did not state which versions those are (except for `compute-runtime` being an older release).

FYI: the most relevant component would have been IGC compiler, as that's the one using SPIRV-translator.

ohsix, 2024-05-20:
> Driver version : 24.09.28717.17 (Linux x64)

> worked on 39, everything is basically a no change rebuild for 40

they're the same package, only rebuilt. as is intel-igc 

https://src.fedoraproject.org/rpms/intel-igc/blob/f39/f/intel-igc.spec
https://src.fedoraproject.org/rpms/intel-igc/blob/f40/f/intel-igc.spec

an assumption changed somewhere (that i'm looking for.) the "1.6" change in spirv-llvm-translator is evidence that the assumption changed

it is often the case when someone has the ability to help themselves that they've already done a lot of investigation and genuinely need very specific advice. knowing it works on ubuntu (intel clearly states, many places, that they support 20 and 22) does not help them.

this basically broke every program that uses opencl in fedora 40, i found out when attempting to use gimp for the first time after the upgrade. the answer to that is not 'use an ubuntu lts'. this wasn't a FTBFS or a crash, a library calls exit(), not at the site the assumption changes, but when use of the bitcode is attempted, and rejected. something changed the precondition both the producer and consumer relied on, and for whatever reason the producer followed suit but the consumer didn't

i'm not looking for evidence that the software worked somewhere. it worked in 39 and it's the same software. there are several presentations and talks about why llvm15 is still being used, and the trouble involved in adapting to newer releases. they're pertinent to this problem and the complexity of the assumptions

i appreciate your help but it was nothing that wasn't known already. this is a last ditch effort to find someone that had a good overview of the whole stack to give me leads to follow up on. there are many moving parts and it isn't just a matter of building them correctly. 'correct' versions of the software contain assumptions that lead one of the libraries to call exit()

i closed the bug because the response changed my expectations of how the information would be seen. unless you know who should look at this, and can contact them, i'm content to wait until someone that knows more fixes it rather than spend a few weeks (... months?) becoming an expert

eero-t, 2024-05-21:
My builds differ a bit from the project builds because I use more components (prebuilds) from the distro than them (LLVM, opencl-clang, llvmspirvlib), so I think there's some extra value.  But I'm using LLVM 14, as that has production support, while support for newer LLVM versions are experimental: https://github.com/intel/intel-graphics-compiler/projects/5

Looking at the Fedora package and comparing the deps to Ubuntu, I see one potential problem. `intel-opencl-clang-devel` has no LLVM version in its name, like the other dependencies.  Is that also built against same LLVM and SPIRV versions as rest of  the components (as it should)?

For answers, it would be better to file this against IGC.  You would need to list versions of _all_ components, preferably also LLVM etc versions those components were _built against_ though.

ohsix, 2024-05-22:
will probably do that, thank you.

was thinking about what i wrote and remembered it's trivial to run the older packages on fedora. short term i can install the f39 version. that also made it clear there were some packages that got bumped that i hadn't looked at

39:
```
# dnf install --releasever=39 intel-opencl
Dependencies resolved.
================================================================================
 Package                     Arch     Version                   Repo       Size
================================================================================
Installing:
 intel-opencl                x86_64   24.09.28717.17-1.fc39     updates   2.4 M
Installing dependencies:
 intel-igc                   x86_64   1.0.16510.1-1.fc39        updates   2.2 M
 intel-igc-libs              x86_64   1.0.16510.1-1.fc39        updates   9.2 M
 intel-opencl-clang          x86_64   15.0.0-8.fc39             updates    12 M
 lld15-libs                  x86_64   15.0.7-4.fc39             fedora    1.4 M
 spirv-llvm15.0-translator   x86_64   15.0.0-4.fc39             updates   826 k
```

40:
```
# dnf install intel-opencl
Last metadata expiration check: 1:42:16 ago on Tue 21 May 2024 09:35:18 PM PDT.
Dependencies resolved.
================================================================================
 Package                     Arch     Version                   Repo       Size
================================================================================
Installing:
 intel-opencl                x86_64   24.09.28717.17-1.fc40     fedora    2.4 M
Installing dependencies:
 intel-igc                   x86_64   1.0.16510.1-1.fc40        fedora    2.1 M
 intel-igc-libs              x86_64   1.0.16510.1-1.fc40        fedora    9.2 M
 intel-opencl-clang          x86_64   15.0.0-10.fc40            fedora     12 M
 lld15-libs                  x86_64   15.0.7-6.fc40             fedora    1.3 M
 spirv-llvm15.0-translator   x86_64   15.0.0-7.fc40             updates   803 k
```

intel-opencl-clang, lld15-libs and spirv-llvm15.0-translator actually got *slightly more* than just a rebuild

will keep you posted

eero-t, 2024-05-22:
> was thinking about what i wrote and remembered it's trivial to run the older packages on fedora. short term i can install the f39 version. that also made it clear there were some packages that got bumped that i hadn't looked at

What about versions for `clpeak` and its (build) dependencies, are those same?

> intel-opencl-clang, lld15-libs and spirv-llvm15.0-translator actually got slightly more than just a rebuild

The upstream versions are still the same, only packaging (patch-level) versions changed.   There's something really broken in those if they would explains the breakage.

frantisekz, 2024-05-23:
@ohsix https://bodhi.fedoraproject.org/updates/FEDORA-2024-042ebc3a54 should fix the issue (I am maintainer of the Intel CL stack in Fedora).

ohsix, 2024-05-24:
excellent! thank you. was here just to say that! haha