intel/compute-runtime #733: ZE_extension_kernel_max_group_size_properties is not reported, even though it's supported

maleadt, 2024-05-15:
I'm using NEO v24.13.29138.7, which includes support for `ZE_extension_kernel_max_group_size_properties` added in https://github.com/intel/compute-runtime/commit/cebedb50c75ceffba4d4cf303ae8f1781376c3d8. However, the extension doesn't seem to be reported by the driver:

```julia
julia> oneL0.extension_properties(driver())
Dict{String, VersionNumber} with 22 entries:
  "ZE_experimental_scheduling_hints"                  => v"1.0.0"
  "ZE_extension_cache_reservation"                    => v"1.0.0"
  "ZE_extension_raytracing"                           => v"1.0.0"
  "ZE_extension_device_luid"                          => v"1.0.0"
  "ZE_extension_memory_free_policies"                 => v"1.0.0"
  "ZE_experimental_event_pool_counter_based"          => v"1.0.0"
  "ZE_extension_pci_properties"                       => v"1.0.0"
  "ZE_extension_device_memory_properties"             => v"1.0.0"
  "ZE_experimental_global_offset"                     => v"1.0.0"
  "ZE_experimental_power_saving_hint"                 => v"1.0.0"
  "ZE_extension_device_ip_version"                    => v"1.0.0"
  "ZE_intel_experimental_event_sync_mode"             => v"1.0.0"
  "ZE_experimental_relaxed_allocation_limits"         => v"1.0.0"
  "ZE_intel_experimental_command_list_memory_sync"    => v"1.0.0"
  "ZE_extension_image_view_planar"                    => v"1.0.0"
  "ZE_intel_experimental_device_module_dp_properties" => v"1.0.0"
  "ZE_extension_memory_compression_hints"             => v"1.0.0"
  "ZE_experimental_rtas_builder"                      => v"1.0.0"
  "ZE_extension_float_atomics"                        => v"1.0.0"
  "ZE_extension_image_view"                           => v"1.0.0"
  "ZE_extension_event_query_kernel_timestamps"        => v"1.0.0"
  "ZE_experimental_module_program"                    => v"1.0.0"
```

This is oneAPI.jl/Julia, but it's basically just calling `zeDriverGetExtensionProperties`. Also, I don't see `ZE_extension_kernel_max_group_size_properties` in https://github.com/intel/compute-runtime/blob/7aceed58ca24772d7f4f6c915d61712a6d85e8dd/level_zero/core/source/driver/driver_handle_imp_helper.cpp

maleadt, 2024-05-16:
If I ignore the extension not being reported by `zeDriverGetExtensionProperties`, querying `ZE_extension_kernel_max_group_size_properties` works fine _except_ when running under `ZE_ENABLE_PARAMETER_VALIDATION=1 ZE_ENABLE_VALIDATION_LAYER=1`. Is this expected?

JablonskiMateusz, 2024-05-16:
Hi @maleadt 
Thanks for catching that, it looks like a gap on our side.

JablonskiMateusz, 2024-07-09:
fix merged https://github.com/intel/compute-runtime/commit/20e22deced540b842187218b6f32ad2fd4a66de9

closed 2024-07-31 by JablonskiMateusz

maleadt, 2024-08-06:
Thanks, verified that it works.