intel/compute-runtime #734: Darktable Hangs with Intel OpenCL Enabled after Update to Kernel 6.8 + compute-runtime

PorcelainMouse, 2024-05-20:
Since kernel 6.8 + compute-runtime, Intel A750 hasn't worked correctly.  With kernel 6.7, BOINC was working well and no problems with Darktable were noticed.  However, immediately after  kernel 6.8, A750 wasn't even detected by BOINC.  After an update to intel-compute-runtime recently, A750 is detected, but performance is much worse, power usage down by 90 % for the same workload.  clpeak hangs.

I just tested Darktable, too, and it completely hangs with Intel OCL enabled, but works with Intel OCL disabled.  I'm assuming this is due to the same underlying performance issue, but I thought I would report it because I couldn't see any other recent issues for darktable.

PorcelainMouse, 2024-05-21:
Awesome.  I saw that, but wasn't sure. I'll try to get back on upstream kernel or at least try 6.9.

ph0b, 2024-05-22:
if you have troubles with the current kernel on your system while waiting for the fix to land, you may work around the issue by using these environment variables:
`NEOReadDebugKeys=1 NodeOrdinal=0` that will force all compute execution to run on the RCS (Render Command Streamer).

eero-t, 2024-06-18:
> Awesome. I saw that, but wasn't sure. I'll try to get back on upstream kernel or at least try 6.9.

As discussed in #726, you will need 6.10 kernel for full (= no perf regression) fix.

eero-t, 2024-07-17:
> > Awesome. I saw that, but wasn't sure. I'll try to get back on upstream kernel or at least try 6.9.
> 
> As discussed in #726, you will need 6.10 kernel for full (= no perf regression) fix.

Which has also been released, few days ago: https://www.phoronix.com/news/Linux-6.10-Released

@PorcelainMouse Please verify / close when convenient.

PorcelainMouse, 2024-07-28:
Yes, well, I'm on kernel 6.9.10 now.  I got excited and thought I got 6.10 today. I re-enabled OCL & Intel OCL, restarted, it did some building of things, then I restarted it twice.  No problems were noticed, seems to run smooth.  But, if 6.10 is indicated, I'll revert back and wait some more. I haven't had time to get back to pulling from upstream kernel. It's just very inconvenient to manage that.

Thanks for the information, though! Much appreciated.

PorcelainMouse, 2024-08-10:
Well, I have kernel 6.10, now, and dark table isn't hanging with Intel OCL enabled.

PorcelainMouse, 2025-11-16:
I can't confirm 100 % because I haven't used Darktable in a while, so I don't have a sense of the performance issues anymore.  I started it just now, and it did a bunch of OpenCL Intel stuff, updating things and whatnot.  It also updated some databases.  In any case, I can again confirm that the program does not hang with Intel OpenCL is enabled.