intel/compute-runtime #737: Unclear whether Atom Gen11 GPU is Supported

PatrickHaecker, 2024-06-13:
According to the [supported platform table](https://github.com/intel/compute-runtime?tab=readme-ov-file#supported-platforms) Level Zero support is unavailable for Intel Atom Processors with Gen11 graphics devices (formerly Elkhart Lake).

However, according to the [FAQ](https://github.com/intel/compute-runtime/blob/master/FAQ.md#how-can-i-check-that-my-specific-device-is-supported-by-the-driver), my device ID 0x4571 is [listed in the code](https://github.com/intel/compute-runtime/blob/master/shared/source/dll/devices/devices_base.inl).

Is Atom Gen11 GPU supported or isn't it?

Initializing the driver fails when executing `env ZE_ENABLE_LOADER_DEBUG_TRACE=1 julia -e "using oneAPI"` with the following output:

```
┌ Error: Failed to initialize oneAPI
│   exception =
│    ZeError: driver is not initialized (code 2013265921, ZE_RESULT_ERROR_UNINITIALIZED)
│    Stacktrace:
│      [1] throw_api_error(res::oneAPI.oneL0._ze_result_t)
│        @ oneAPI.oneL0 ~/.julia/packages/oneAPI/1GTs3/lib/level-zero/libze.jl:8
│      [2] check
│        @ ~/.julia/packages/oneAPI/1GTs3/lib/level-zero/libze.jl:19 [inlined]
│      [3] zeInit
│        @ ~/.julia/packages/oneAPI/1GTs3/lib/utils/call.jl:24 [inlined]
│      [4] __init__()
│        @ oneAPI.oneL0 ~/.julia/packages/oneAPI/1GTs3/lib/level-zero/oneL0.jl:131
│      [5] run_module_init(mod::Module, i::Int64)
│        @ Base ./loading.jl:1134
│      [6] register_restored_modules(sv::Core.SimpleVector, pkg::Base.PkgId, path::String)
│        @ Base ./loading.jl:1122
│      [7] _include_from_serialized(pkg::Base.PkgId, path::String, ocachepath::String, depmods::Vector{Any})
│        @ Base ./loading.jl:1067
│      [8] _require_search_from_serialized(pkg::Base.PkgId, sourcepath::String, build_id::UInt128)
│        @ Base ./loading.jl:1581
│      [9] _require(pkg::Base.PkgId, env::String)
│        @ Base ./loading.jl:1938
│     [10] __require_prelocked(uuidkey::Base.PkgId, env::String)
│        @ Base ./loading.jl:1812
│     [11] #invoke_in_world#3
│        @ ./essentials.jl:926 [inlined]
│     [12] invoke_in_world
│        @ ./essentials.jl:923 [inlined]
│     [13] _require_prelocked(uuidkey::Base.PkgId, env::String)
│        @ Base ./loading.jl:1803
│     [14] macro expansion
│        @ ./loading.jl:1790 [inlined]
│     [15] macro expansion
│        @ ./lock.jl:267 [inlined]
│     [16] __require(into::Module, mod::Symbol)
│        @ Base ./loading.jl:1753
│     [17] #invoke_in_world#3
│        @ ./essentials.jl:926 [inlined]
│     [18] invoke_in_world
│        @ ./essentials.jl:923 [inlined]
│     [19] require(into::Module, mod::Symbol)
│        @ Base ./loading.jl:1746
│     [20] eval
│        @ ./boot.jl:385 [inlined]
│     [21] exec_options(opts::Base.JLOptions)
│        @ Base ./client.jl:291
│     [22] _start()
│        @ Base ./client.jl:552
└ @ oneAPI.oneL0 ~/.julia/packages/oneAPI/1GTs3/lib/level-zero/oneL0.jl:134
```

This is the output of `lspci -vv -s 0:2`:

```
00:02.0 VGA compatible controller: Intel Corporation Device 4571 (rev 01) (prog-if 00 [VGA controller])
        Subsystem: Intel Corporation Device 7270
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 176
        Region 0: Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 4000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: <access denied>
        Kernel driver in use: i915
        Kernel modules: i915
```

This is the output of `lscpu`

```
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  4
  On-line CPU(s) list:   0-3
Vendor ID:               GenuineIntel
  Model name:            Intel Atom(R) x6425RE Processor @ 1.90GHz
    CPU family:          6
    Model:               150
    Thread(s) per core:  1
    Core(s) per socket:  4
    Socket(s):           1
    Stepping:            1
    CPU max MHz:         1900,0000
    CPU min MHz:         800,0000
    BogoMIPS:            3763.20
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop
                         _tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg cx16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand lahf_lm 3dnowprefetch cpuid_fa
                         ult epb cat_l2 cdp_l2 ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust smep erms rdt_a rdseed smap clflushopt clwb intel_pt sha_ni xsaveopt xsavec xgetbv1 xsave
                         s split_lock_detect dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req umip waitpkg gfni rdpid movdiri movdir64b md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   128 KiB (4 instances)
  L1i:                   128 KiB (4 instances)
  L2:                    1,5 MiB (1 instance)
  L3:                    4 MiB (1 instance)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-3
Vulnerabilities:         
  Gather data sampling:  Not affected
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Mitigation; Clear CPU buffers; SMT disabled
  Retbleed:              Not affected
  Spec rstack overflow:  Not affected
  Spec store bypass:     Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:            Mitigation; Enhanced IBRS, IBPB conditional, RSB filling, PBRSB-eIBRS Not affected
  Srbds:                 Vulnerable: No microcode
  Tsx async abort:       Not affected
```

JablonskiMateusz, 2024-06-18:
Hi @PatrickHaecker 

Intel Atom is not supported by Level Zero. The list you mentioned https://github.com/intel/compute-runtime/blob/master/shared/source/dll/devices/devices_base.inl is for shared code between OpenCL and Level Zero. The L0 support is defined here: https://github.com/intel/compute-runtime/blob/master/shared/source/gen11/hw_info_ehl.cpp#L73

PatrickHaecker, 2024-06-18:
Hi @JablonskiMateusz,

thanks a lot for the clarification.

Is there a time plan when (or whether at all) Gen11 Atom will have L0 support?

The company I am working for might have a business need for that support in the future (currently one of multiple options). Is there anyone to talk about such a support if development could possibly be paid for?

AdamCetnerowski, 2024-06-24:
Please reach out to Michael Drozd ([michael.drozd@intel.com](mailto:michael.drozd@intel.com)) who will work with you on the request

PatrickHaecker, 2024-06-25:
Thanks for the contact. I'll reach out to Michael Drozd.

closed 2024-06-25 by PatrickHaecker