intel/compute-runtime #738: Build fail with 24.22.29735.x  (=latest) tags

eero-t, 2024-06-18:
Building latest `compute-runtime` tag (`24.22.29735.15`) fails:
```
$ cmake -LH -Wno-dev -G Ninja  -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_BUILD_TYPE=Release -DSUPPORT_GEN8=0 -DSUPPORT_GEN9=1 -DSUPPORT_GEN11=0 -DSUPPORT_TGLLP=1 -DSUPPORT_DG1=1 -DSUPPORT_XE_HP_SDV=1-DSUPPORT_DG2=1 -DSUPPORT_PVC=1 -DNEO_ENABLE_i915_PRELIM_DETECTION=TRUE -DNEO_DISABLE_LD_GOLD=1 -DDO_NOT_RUN_AUB_TESTS=1 -DDONT_CARE_OF_VIRTUALS=1
$ ninja
...
#21 291.5 /usr/bin/c++ @level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/driver/driver_handle_imp.cpp.o.rsp -MD -MT level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/driver/driver_handle_imp.cpp.o -MF level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/driver/driver_handle_imp.cpp.o.d -o level_zero/CMakeFiles/ze_intel_gpu_lib.dir/core/source/driver/driver_handle_imp.cpp.o -c /home/nobody/source/compute-runtime/level_zero/core/source/driver/driver_handle_imp.cpp
#21 291.5 /home/nobody/source/compute-runtime/level_zero/core/source/driver/driver_handle_imp.cpp: In member function 'virtual ze_result_t L0::DriverHandleImp::getProperties(ze_driver_properties_t*)':
#21 291.5 /home/nobody/source/compute-runtime/level_zero/core/source/driver/driver_handle_imp.cpp:118:70: error: expected primary-expression before ')' token
#21 291.5   118 |     uint32_t versionHotfix = static_cast<uint32_t>(NEO_VERSION_HOTFIX);
#21 291.5       |                                                                      ^
```

Previous `24.17.29377.x` tag series builds fine.

=> Culprit seems to be broken 56847274fab commit.

JablonskiMateusz, 2024-07-08:
the commit was reverted on 24.22 release branch at 24.22.29735.20 
Please confirm if it works fine

eero-t, 2024-07-08:
Buildling `24.22.29735.27` (latest in `24.22.29735.x` tag series) works -> closing.

closed 2024-07-08 by eero-t