intel/compute-runtime #740: Cannot create Virtual Memory on Synology DSM7 (kernel 4.x)

chrisdecker08, 2024-06-23:
After updating from 23.05.25593.11 to 24.17.29377.6 tonemapping via ffmpeg no longer succeeds.  Tuning on PrintDebugMessages results in the following:

> WARNING: Topology query failed!
> INFO: System Info query failed!
> WARNING: Failed to request OCL Turbo Boost
> WARNING: Failed to query memory info
> WARNING: Failed to query engine info
> WARNING: Topology query failed!
> NEO_CACHE_PERSISTENT is enabled. Cache is located in: /var/packages/PlexMediaServer/home/.cache/neo_compiler_cache
> 
> INFO: Cannot create Virtual Memory at memory bank 0x1 info present 0  return code -1
> [AVHWDeviceContext @ 0x7f8348131a80] Failed to get number of OpenCL platforms: -1001.
>     Last message repeated 1 times
> [Parsed_hwmap_3 @ 0x7f83480f7040] Failed to created derived device context: -19.
> [Parsed_hwmap_3 @ 0x7f83480f7040] Failed to configure output pad on Parsed_hwmap_3
> Error reinitializing filters!
> Failed to inject frame into filter network: No such device
> Error while processing the decoded data for stream #0:0

JablonskiMateusz, 2024-06-27:
Hi @chrisdecker08  
could you share more details about your setup?

chrisdecker08, 2024-06-27:
We package and ship ffmpeg/icr for a variety of linux OSs/NAS devices.  While I don't have a setup to reproduce this myself many of our users have reported the breakage on Synology's DSM7 OS for NAS devices.  I am working with one such user to try and identify the unerlying issue and so far it is looking to be DRM related.  I will let you know when i've identified the underlying issue.

chrisdecker08, 2024-07-02:
it looks like [this](https://github.com/intel/compute-runtime/commit/d8f99161dda51bae829e5260b37cf6b8a35ac5e7) change is causing the failure for these devices due to createDrmVirtualMemory returning an error on line 55 of os_context_linux.cpp

HoppeMateusz, 2024-07-08:
Hi @chrisdecker08 

Can you help with identifying which ioctl fails ? 

you may use this debug variable : PrintIoctlEntries=1 
NEOReadDebugKeys=1 PrintDebugMessages=1  PrintIoctlEntries=1    ./test ....

I suspect VM CREATE ioctl may be failing:
DRM_IOCTL_I915_GEM_VM_CREATE

chrisdecker08, 2024-07-08:
I would not be surprised if Synology shipped a customized or old kernel and that is causing the problems.  It should be noted that if I remove os_context_linux.cpp lines 63-65 tone mapping does apply correctly.

here are the ioctl results:

```
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_CHIPSET_ID, output value: 23173, retCode: 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_REVISION, output value: 11, retCode: 0
IOCTL DRM_IOCTL_I915_GEM_CREATE_EXT called
IOCTL DRM_IOCTL_I915_GEM_CREATE_EXT returns -1, errno 22(Invalid argument)
IOCTL DRM_IOCTL_I915_REG_READ called
IOCTL DRM_IOCTL_I915_REG_READ returns 0
IOCTL DRM_IOCTL_I915_QUERY called
IOCTL DRM_IOCTL_I915_QUERY returns -1, errno 22(Invalid argument)
WARNING: Topology query failed!
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_EU_TOTAL, output value: 12, retCode: 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_SUBSLICE_TOTAL, output value: 2, retCode: 0
IOCTL DRM_IOCTL_I915_QUERY called
IOCTL DRM_IOCTL_I915_QUERY returns -1, errno 22(Invalid argument)
INFO: System Info query failed!
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM returns -1, errno 22(Invalid argument)
WARNING: Failed to request OCL Turbo Boost
IOCTL DRM_IOCTL_I915_QUERY called
IOCTL DRM_IOCTL_I915_QUERY returns -1, errno 22(Invalid argument)
WARNING: Failed to query memory info
IOCTL DRM_IOCTL_I915_QUERY called
IOCTL DRM_IOCTL_I915_QUERY returns -1, errno 22(Invalid argument)
WARNING: Failed to query engine info
IOCTL DRM_IOCTL_I915_QUERY called
IOCTL DRM_IOCTL_I915_QUERY returns -1, errno 22(Invalid argument)
WARNING: Topology query failed!
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_EU_TOTAL, output value: 12, retCode: 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_SUBSLICE_TOTAL, output value: 2, retCode: 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_HAS_POOLED_EU, output value: 0, retCode: 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns -1, errno 22(Invalid argument)
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns -1, errno 22(Invalid argument)
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_HAS_SCHEDULER, output value: 7, retCode: 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns 0
IOCTL DRM_IOCTL_I915_GEM_USERPTR called
IOCTL DRM_IOCTL_I915_GEM_USERPTR returns 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns 0
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM called
IOCTL DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM returns 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_CS_TIMESTAMP_FREQUENCY, output value: 19200000, retCode: 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_CS_TIMESTAMP_FREQUENCY, output value: 19200000, retCode: 0
```

HoppeMateusz, 2024-07-09:
are those results from run that fails ? 

i cannot see failing GEM VM CREATE ioctl - while previously there was error message printed after that ioclt failed:

    auto ret = ioctlHelper->ioctl(DrmIoctl::gemVmCreate, &ctl);

    if (ret == 0) {
        drmVmId = ctl.vmId;
        if (ctl.vmId == 0) {
            // 0 is reserved for invalid/unassigned ppgtt
            return -1;
        }
    } else {
        printDebugString(debugManager.flags.PrintDebugMessages.get(), stderr,
                         "INFO: Cannot create Virtual Memory at memory bank 0x%x info present %d  return code %d\n",
                         memoryBank, memoryInfo != nullptr, ret);
    }

chrisdecker08, 2024-07-09:
It was from the run that failed, I just missed the last couple lines from the log

```
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_CS_TIMESTAMP_FREQUENCY, output value: 19200000, retCode: 0
IOCTL DRM_IOCTL_I915_GETPARAM called
IOCTL DRM_IOCTL_I915_GETPARAM returns 0
DRM_IOCTL_I915_GETPARAM: param: I915_PARAM_CS_TIMESTAMP_FREQUENCY, output value: 19200000, retCode: 0
NEO_CACHE_PERSISTENT is enabled. Cache is located in: /root/.cache/neo_compiler_cache

IOCTL DRM_IOCTL_I915_GEM_VM_CREATE called
IOCTL DRM_IOCTL_I915_GEM_VM_CREATE returns -1, errno 22(Invalid argument)
INFO: Cannot create Virtual Memory at memory bank 0x1 info present 0  return code -1
IOCTL DRM_IOCTL_GEM_CLOSE called
IOCTL DRM_IOCTL_GEM_CLOSE returns 0
[AVHWDeviceContext @ 0x7fe7bbcf36c0] Failed to get number of OpenCL platforms: -1001.
    Last message repeated 1 times
[Parsed_hwmap_3 @ 0x7fe7bbcbd040] Failed to created derived device context: -19.
[Parsed_hwmap_3 @ 0x7fe7bbcbd040] Failed to configure output pad on Parsed_hwmap_3
```

HoppeMateusz, 2024-07-09:
Thanks, 
so as i suspected, VM CREATE ioctl fails. 

I am preparing fix for that, should be available shortly (today or tomorrow)

HoppeMateusz, 2024-07-10:
Here is the change that should fix the problem: https://github.com/intel/compute-runtime/commit/c660784df2f145808454fffc6883ba8afb3ef24c

chrisdecker08, 2024-07-23:
thanks!  this works perfectly