intel/compute-runtime #743: clSharedMemAllocINTEL and clMemBlockingFreeINTEL have higher overhead on windows than linux

xiang1guo, 2024-07-10:
Hi, team

Recently, my program showed that Windows has a high host overhead against Linux. I used cliloader to profile the cl calls and finally found that these 2 functions(clSharedMemAllocINTEL and clMemBlockingFreeINTEL) cost too much time on Windows than on Linux. 

My question is: Is it expected? Would you happen to have any idea about this? Thanks!

smorek-intel, 2024-07-15:
Hi @xiang1guo
Linux i915 KMD has an optimization to reuse freed gfx allocations. This is not the case on Windows.
Best approach here is to reduce amount of allocations by user-mode memory pool.

xiang1guo, 2024-07-16:
> Hi @xiang1guo Linux i915 KMD has an optimization to reuse freed gfx allocations. This is not the case on Windows. Best approach here is to reduce amount of allocations by user-mode memory pool.

Thanks Morek, thanks for the reply and the information you provide. I finally understand the root-cause of the regression on Windows compared to Linux. I am going to close the issue and thanks again for your support!

closed 2024-07-16 by xiang1guo