intel/compute-runtime #752: FTBFS with gcc-14

tjaalton, 2024-08-06:
building 24.26.30049.8 fails on Debian unstable:

```
In file included from /usr/include/c++/14/vector:66,
                 from /usr/include/c++/14/functional:64,
                 from /<<PKGBUILDDIR>>/shared/source/helpers/aligned_memory.h:14,
                 from /<<PKGBUILDDIR>>/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_alloca
tion.cpp:9:
In member function ‘std::vector<NEO::BufferObject*, std::allocator<NEO::BufferObject*> >::operator[](unsigned long)
’,
    inlined from ‘StackVec<NEO::BufferObject*, 4ul, unsigned char>::operator[](unsigned long)’ at /<<PKGBUILDDIR>>/
shared/source/utilities/stackvec.h:264:33,
    inlined from ‘NEO::DrmAllocation::DrmAllocation(unsigned int, unsigned long, NEO::AllocationType, NEO::BufferOb
ject*, void*, unsigned long, unsigned long, NEO::MemoryPool)’ at /<<PKGBUILDDIR>>/shared/source/os_interface/linux/
drm_allocation.h:47:24,
    inlined from ‘NEO::DrmMemoryManager::createMultiHostAllocation(NEO::AllocationData const&)’ at /<<PKGBUILDDIR>>
/shared/source/os_interface/linux/drm_memory_manager_create_multi_host_allocation.cpp:47:133:
/usr/include/c++/14/bits/stl_vector.h:1131:32: error: ‘MEM[(struct vector *)_48 + 1528B].D.179414._M_impl.D.178750._M_start’ may be used uninitialized [-Werror=maybe-uninitialized]
 1131 |         return *(this->_M_impl._M_start + __n);
      |                  ~~~~~~~~~~~~~~^~~~~~~~
```

JablonskiMateusz, 2024-08-23:
Hi @tjaalton 

Thanks for reporting the issue. 
I already prepared fix for gcc14 compilation: https://github.com/intel/compute-runtime/commit/70bb654c15d043fb2445a9a3742e45d4b5579ccf 
This one is for default compilation with disabled lto.
Remaining fixes will be available soon

JablonskiMateusz, 2024-08-26:
Two remaining fixes: https://github.com/intel/compute-runtime/commit/dbc8baddc1f1fd1fe39dcbce661c81a3beb6e2f5 https://github.com/intel/compute-runtime/commit/3e5dec27ef8bb637fd4491dcf2df01bec0af05ff

Now you should be able to build NEO with gcc14 without any issue

tjaalton, 2024-08-26:
yep, builds with these, thanks!

closed 2024-08-26 by tjaalton