intel/compute-runtime #753: how to install on intel clear linux

MajidAbdelilah, 2024-08-09:
how to install the compute-runtime on intel clear linux
there is no package of it in intels own linux distro

eero-t, 2024-08-19:
> there is no package of it in intels own linux distro

ClearLinux dropped `compute-runtime` packages from it some time ago.

Reasons:
* Distros build drivers using system compiler versions
* ClearLinux tracks latest LLVM version and does _not_ include support for building code with older LLVM versions [1]
* `compute-runtime` support for latest LLVM versions comes with considerable delay (too much for ClearLinux)

> how to install the compute-runtime on intel clear linux

This is not really supported, but...

I've successfully used driver libs built for Ubuntu on ClearLinux, as long as they do link to LLVM dynamically.  And `compute-runtime` releases here (and in Intel's own driver repos) link LLVM statically, so latter is not a problem.

Have you tried just extracting contents from the latest (*.deb) release packages here to (a directory in linker path in) Clear Linux?

-----

[1] Mainstream distros such as Debian, Ubuntu & Fedora support multiple LLVM versions so this is less of a problem for them.

MajidAbdelilah, 2024-08-19:
Sorry for the hustle. I have deleted clear linux and installed arch just because of compute runtime and oneapi.
This is the third time i swiched from clear to another distro and if i learned anything. Clear linux is not for developers.

MajidAbdelilah, 2024-08-19:
Thank you for your efforts

closed 2024-09-25 by JablonskiMateusz