intel/compute-runtime #754: segfault in clIcdGetPlatformIDsKHR

chrisdecker08, 2024-08-11:
I have seen reports of some of our users segfaulting when utilizing tonemapping functionality in ffmpeg.  The users are inconsistent (2 users with the same distro/version and graphics cards in the same family are seeing 2 different results) but the same user will consistently run fine or segfault.  I was able to get the following stacktrace for the crash.

```
* thread #1, name = 'Plex Transcoder', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
  * frame #0: 0x00007ffff28897df libigdrcl.so`NEO::GfxCoreHelperHw<NEO::XeHpgCoreFamily>::getGpgpuEngineInstances(this=0x00007ffff3c6a270, rootDeviceEnvironment=0x00007ffff7e61160) const at gfx_core_helper_xehp_and_later.inl:81:101
    frame #1: 0x00007ffff2a73777 libigdrcl.so`NEO::ExecutionEnvironment::calculateMaxOsContextCount(this=0x00007ffff14758d0) at execution_environment.cpp:95:47
    frame #2: 0x00007ffff2afc674 libigdrcl.so`NEO::DeviceFactory::prepareDeviceEnvironments(executionEnvironment=0x00007ffff14758d0) at device_factory.cpp:201:26
    frame #3: 0x00007ffff29b2107 libigdrcl.so`NEO::prepareDeviceEnvironmentsImpl(executionEnvironment=0x00007ffff14758d0) at create_command_stream_impl.cpp:57:16
    frame #4: 0x00007ffff21fb7b6 libigdrcl.so`NEO::prepareDeviceEnvironments(executionEnvironment=0x00007ffff14758d0, osPciPath="", rootDeviceIndex=0) at get_devices.cpp:20:23
    frame #5: 0x00007ffff21fbbb2 libigdrcl.so`NEO::prepareDeviceEnvironments(executionEnvironment=0x00007ffff14758d0) at get_devices.cpp:53:12
    frame #6: 0x00007ffff2afcafd libigdrcl.so`NEO::DeviceFactory::createDevices(executionEnvironment=0x00007ffff14758d0) at device_factory.cpp:242:10
    frame #7: 0x00007ffff21fdf43 libigdrcl.so`::clGetPlatformIDs(numEntries=0, platforms=0x0000000000000000, numPlatforms=0x00007fffffffa9cc) at api.cpp:101:31
    frame #8: 0x00007ffff21ff196 libigdrcl.so`::clIcdGetPlatformIDsKHR(numEntries=0, platforms=0x0000000000000000, numPlatforms=0x00007fffffffa9cc) at api.cpp:148:14
    frame #9: 0x00007ffff6ade2a0 libavutil.so.57`___lldb_unnamed_symbol1393 + 162
    frame #10: 0x00007ffff6ae097f libavutil.so.57`___lldb_unnamed_symbol1553 + 445
    frame #11: 0x00007ffff6ae0619 libavutil.so.57`___lldb_unnamed_symbol1550 + 352
    frame #12: 0x00007ffff6ae048c libavutil.so.57`___lldb_unnamed_symbol1549 + 50
```

chrisdecker08, 2024-08-11:
I actually tracked this down to the fact we disable AIL and submitted a fix here: https://github.com/intel/compute-runtime/pull/755

JablonskiMateusz, 2024-08-20:
Hi @chrisdecker08 
Thanks for reporting the issue. Disabling AIL may in fact impact here as we missed some checks for nullptr after adding debug key to disable AIL. 
I'll provide wider fix with corresponding ULTs

chrisdecker08, 2024-08-20:
@JablonskiMateusz can you please tag me in the wider fix so I can apply the patch to our version?

closed 2024-08-20 by VPG-SWE-Github

JablonskiMateusz, 2024-08-21:
@chrisdecker08 fix merged in https://github.com/intel/compute-runtime/commit/f617093a6a0154e012ee758aaa3fff6e9e0ac885