intel/compute-runtime #756: You are trying to dlopen a libur_adapter_level_zero.so.0 shared library with RTLD_DEEPBIND flag which is incompatible with sanitizer runtime

MajidAbdelilah, 2024-08-13:
```dpcpp host_vs_device_code.cpp -g -fsanitize=thread```
```./a.out```
```==6411==You are trying to dlopen a libur_adapter_level_zero.so.0 shared library with RTLD_DEEPBIND flag which is incompatible with sanitizer runtime (see https://github.com/google/sanitizers/issues/611 for details). If you want to run libur_adapter_level_zero.so.0 library under sanitizers please remove RTLD_DEEPBIND from dlopen flags.```

i am using arch linux with the dpcpp installed via pacman
also the package for dpcpp in arch linux doesnt have the ```/opt/intel/oneapi/compiler/latest/lib/clang/18/lib/linux/``` folder so i have copied it from my system clang folder(from ```/usr/lib/clang/18/lib/linux/```)
befor cpying it when i try to use a sanitizer the compiler complain about not finding a library inside that forlder(```/opt/intel/oneapi/compiler/latest/lib/clang/18/lib/linux/xxxxxxx```)
is there a fix to this

JablonskiMateusz, 2024-08-14:
@MajidAbdelilah  is it related to libze_intel_gpu.so.1 ? please capture which library is using deepbind

MajidAbdelilah, 2024-08-14:
Neverminde i installed intel oneapi base toolkit and everything works fine now. Its an arch package issue

closed 2024-08-23 by JablonskiMateusz