intel/compute-runtime #757: Q: Possible OOM reported for Host-allocations (L0)

abagusetty, 2024-08-16:
An application reported this error on Aurora upon using `zeMallocHost` or `sycl::malloc_host`

```bash
Abort was called at 80 line in file:
/home/ubit/rpmbuild/BUILD/intel-compute-runtime-24.22.29735.27/shared/source/command_stream/linear_stream.h
Aborted
```
Any pointers on this one. 
Seems like OOM issue: https://github.com/intel/compute-runtime/blob/master/shared/source/command_stream/linear_stream.h#L80

Can someone confirm this

JablonskiMateusz, 2024-08-20:
@abagusetty  could you provide callstack of the issue?

kgibala, 2025-10-15:
Hi @abagusetty ,

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!