intel/compute-runtime #758: cross-compile against Synology toolchain?

th0ma7, 2024-08-24:
### Background
Long-story short, I'm maintaining the ffmpeg package part of SynoCommunity.  I'm trying to enable opencl within ffmpeg.  Our build framework called spksrc allows cross-compiling using synology linux toolchains for various archs.  In this specific case, obviously, I'm cross-compiling for x86_64 intel arch, from a debian 12 x86_64 container.

More info available at: https://github.com/SynoCommunity/spksrc
Current PR: https://github.com/SynoCommunity/spksrc/pull/6166

---

## Issue
We've been supporting VA-API since quite a while, part of current PR I was able to cross-compile sucesfully IGC.  When trying to cross-compile intel-compute-runtime it ends-up strugling with:
```
[ 11%] ^[[32mBuilding CXX object shared/source/CMakeFiles/neo_shared.dir/xe_hpg_core/windows/gmm_callbacks_xe_hpg_core.cpp.o^[[0m
gmake[5]: Entering directory '/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/compute-runtime-24.26.30049.6/build'
[ 11%] ^[[34m^[[1mGenerating ../../../../bin/built_ins/x64/spirv/fill_image3d.spv^[[0m
Error! Loading of FCL library has failed! Filename: libigdfcl.so.1
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
Command was: /home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/compute-runtime-24.26.30049.6/build/bin/ocloc -q -spv_only -file /home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/compute-runtime-24.26.30049.6/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/compute-runtime-24.26.30049.6/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
gmake[5]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make:220: bin/built_ins/x64/spirv/fill_image3d.spv] Error 250
gmake[5]: Leaving directory '/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/compute-runtime-24.26.30049.6/build'
gmake[4]: *** [CMakeFiles/Makefile2:6523: shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/all] Error 2
gmake[4]: *** Waiting for unfinished jobs....
```
But in reality there is no way this can work properly as it is trying to load a `target` shared library.  Should I look into finding a host pre-compiled `ocloc` that it could is part of the build process in order to avoid this?  Or any other alternatives?

Much thanx in advance!

JablonskiMateusz, 2024-08-26:
Hi @th0ma7 
from the log you shared it looks like libigdfcl.so.1 couldn't be loaded.
Since build-time IGC dependency is needed only to provide precompiled kernels within driver and the compilation is performed on host, I think you should use host-based IGC libraries, instead of target-based.

th0ma7, 2024-08-26:
I'll give this a shot but this will extend the build duration by quite much... sigh.

So guessing I'd have to build an version matching host-based IGC, then, when building cross-compiled compute-runtime adding a LD_LIBRARY_PATH to point to my host IGC so it loads proper library from there in order to build precompiled kernels?

JablonskiMateusz, 2024-08-26:
btw, please share log from:
`ldd -r libigdfcl.so.1`

th0ma7, 2024-08-26:
@JablonskiMateusz sure, here it is:
```
~/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime/lib$ ldd -r libigdfcl.so.1
	linux-vdso.so.1 (0x00007ffeba84c000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x000078e8348c1000)
	libopencl-clang.so.14 => not found
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x000078e8348bc000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x000078e8348b7000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x000078e83487d000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x000078e833e00000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000078e83479c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x000078e83477c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000078e83401f000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x000078e834749000)
	/lib64/ld-linux-x86-64.so.2 (0x000078e8348d1000)
undefined symbol: Compile	(./libigdfcl.so.1)
```
Now as for `libopencl-clang.so.14`:
```
~/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime/lib$ ldd -r libopencl-clang.so.14
	linux-vdso.so.1 (0x00007ffe76988000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007b1db83cb000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007b1db83c6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007b1db83c1000)
	libncursesw.so.6 => /lib/x86_64-linux-gnu/libncursesw.so.6 (0x00007b1db8387000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007b1db1000000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007b1db82a6000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007b1db8286000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007b1db121f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007b1db83db000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007b1db8253000)
```
But really this actually relates to a rpath for actual target installation directory (unexisting while cross-compiling), it just hapens that the debian 12 container provides most of the same libraries as well:
```
$ libtree libigdfcl.so.1
libigdfcl.so.1 
├── librt.so.1 [ld.so.conf]
├── libpthread.so.0 [ld.so.conf]
├── libncursesw.so.6 [ld.so.conf]
│   └── libtinfo.so.6 [ld.so.conf]
└── libopencl-clang.so.14 not found
    ┊ Paths considered in this order:
    ┊ 1. rpath:
    ┊    depth 1
    ┊    /usr/local/compute-runtime/lib
    ┊ 2. LD_LIBRARY_PATH was not set
    ┊ 3. runpath was not set
    ┊ 4. ld config files:
    ┊    /usr/lib/x86_64-linux-gnu/libfakeroot
    ┊    /usr/local/lib
    ┊    /usr/local/lib/x86_64-linux-gnu
    ┊    /lib/x86_64-linux-gnu
    ┊    /usr/lib/x86_64-linux-gnu
    ┊    /lib32
    ┊    /usr/lib32
    ┊    /libx32
    ┊    /usr/libx32
    ┊ 5. Standard paths:
    ┊    /lib
    ┊    /lib64
    ┊    /usr/lib
    ┊    /usr/lib64
Error [libigdfcl.so.1]: Not all dependencies were found
```

JablonskiMateusz, 2024-08-26:
you may try to setup LD_LIBRARY_PATH pointing to the directory (`~/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime/lib`)

Besides, as a case-study could you share:
- cmake command
- cmake output log
- verbosed make output log 

?

th0ma7, 2024-08-26:
> Besides, as a case-study could you share:
> * cmake command
> * cmake output log

Here is the log, i've trimmed out the dependency build and only left compute-runtime:  [build-x64-7.1-default.log](https://github.com/user-attachments/files/16750679/build-x64-7.1-default.log)

Error being:
```
Error! Loading of FCL library has failed! Filename: libigdfcl.so.1
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
```

> you may try to setup LD_LIBRARY_PATH pointing to the directory (`~/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime/lib`)

I had manually tried it with a `LD_PRELOAD=libA;libB` and the error message changed, just like in the log below:
```
Error! Loading of FCL library has failed! Filename: libigdfcl.so.1
Error! FCL initialization failure. Error code = -6
Build failed with error code: -6
```
I figured, as libraries I'm loading have been linked against the synology toolchain sysroot which isn't of the same version as the host (i.e. libc and al) it may be an issue, leading to the need of building a host based IGC so libraries can be properly loaded (not tried yet).  Resulting logs:  [build-x64-7.1-LD_PRELOAD.log](https://github.com/user-attachments/files/16750672/build-x64-7.1-LD_PRELOAD.log)

> * verbosed make output log

I had to add `-DCMAKE_VERBOSE_MAKEFILE='ON'` and `--trace-expand` (hopefuly it suffice?)  Otherwise, same as above, with `LD_PRELOAD`, but verbose mode on, compressed as otherwise huge: [build-x64-7.1-LD_PRELOAD+debug.log.gz](https://github.com/user-attachments/files/16750771/build-x64-7.1-LD_PRELOAD%2Bdebug.log.gz)

JablonskiMateusz, 2024-08-27:
It looks like IGC_DIR is not set correctly.
Please change it from:
`-DIGC_DIR=/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime/include`
to
`-DIGC_DIR=/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/compute-runtime`

th0ma7, 2024-08-27:
Aaahh! that fixed it!  Thnx a milion!  got to 100% build complete :rocket: 

Only issue left (AFAIK) that I noticed is the installation of `/etc/OpenCL/vendors/intel.icd` which didn't follow the installation prefix as it should have been under `/usr/local/compute-runtime/etc/OpenCL/vendors/intel.icd` .  Unless the file isn't mandatory and I can just skip packaging it?

th0ma7, 2024-08-27:
Sorry to further bother, I thought I had it all nailed down.  Indeed intel-compute-runtime now builds succesfully (again much thnx!) - unrelated: I added a workaround to install `intel.icd` under the prefix directory.

My following question is, what's next?  I built `ocl-icd` along with `clinfo` and ffmpeg did detected OpenCL through `libOpenCL.so.1.0.0` provided by `ocl-icd`.  Although there's seems to be missing the glue between libOpenCL and compute-runtime.

Here's the driver status:
```
# ./vainfo
Trying display: drm
libva info: VA-API version 1.22.0
libva info: Trying to open /var/packages/ffmpeg6/target/lib/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.2.5 (4c70d6b2)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD

# ./clinfo 
Number of platforms                               0

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.3.2
  ICD loader Profile                              OpenCL 3.0
```

JablonskiMateusz, 2024-08-27:
do you setup DESTDIR when calling `make install`? 
the icd file should be installed in `$DESTDIR/etc/OpenCL/vendors/intel.icd`

anyway, the location of etc is required by OCL loader: https://registry.khronos.org/OpenCL/sdk/3.0/docs/man/html/cl_khr_icd.html#_icd_loader_vendor_enumeration_on_linux

JablonskiMateusz, 2024-08-27:
>  ./clinfo 
Number of platforms                               0
ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.3.2
  ICD loader Profile                              OpenCL 3.0

please run `strace -o strace.log clinfo` and share strace.log file

JablonskiMateusz, 2024-09-26:
@th0ma7 since the issue seems to be resolved, can we close the issue?

th0ma7, 2024-09-27:
actually it doesn't... the DESTDIR did not solved things and I still have to manually move the `intel.icd` at post-install.

JablonskiMateusz, 2024-09-27:
in the log you have 
`DESTDIR=/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install` 
and the intel.icd is installed in 
`/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/etc/OpenCL/vendors/intel.icd` 
which is `$DESTDIR/etc/OpenCL/vendors/intel.icd` as I mentioned earlier

th0ma7, 2024-09-30:
Exactly, it ends-up going into:
```
/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/etc/OpenCL/vendors/
```
instead of `$(DESTDIR)/$(INSTALL_PREFIX)`
```
/home/spksrc/ffmpeg-opencl-140/spksrc/cross/intel-compute-runtime/work-x64-7.1/install/usr/local/intel-compute-runtime/etc/OpenCL/vendors/
```
The output above is when testing the compilation through the `cross/` directory.  Below is the normal type of install when building a `spk` package from the `spk/` directory, but sime concept applies:
- destdir path: `/home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install`
- install prefix (also similar to rpath): `/var/packages/synocli-videodriver/target`

And from my logs I see:
```
===>  - CMake install
cd /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/compute-runtime-24.31.30508.7 && env -u AR -u AS -u CC -u CPP -u CXX -u LD -u NM -u OBJDUMP -u RANLIB -u READELF -u STRIP -u CFLAGS -u CPPFLAGS -u CXXFLAGS -u LDFLAGS  PKG_CONFIG_LIBDIR=/home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install//var/packages/synocli-videodriver/target/lib/pkgconfig WORK_DIR=/home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1 INSTALL_PREFIX=/var/packages/synocli-videodriver/target TC_WORK_DIR=/home/spksrc/ffmpeg7/spksrc/toolchain/syno-x64-7.1/work TC_GCC=$(eval $(echo /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/../../../toolchain/syno-x64-7.1/work/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu-gcc -dumpversion) 2>/dev/null || true) TC_GLIBC=2.26 TC_KERNEL=4.4.180+ LD_LIBRARY_PATH=/home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install/var/packages/synocli-videodriver/target/lib DESTDIR=/home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install cmake --install /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/compute-runtime-24.31.30508.7/build
-- Install configuration: "Release"
-- Installing: /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install/etc/OpenCL/vendors/intel.icd
-- Installing: /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install/var/packages/synocli-videodriver/target/bin/ocloc-24.31
-- Installing: /home/spksrc/ffmpeg7/spksrc/spk/synocli-videodriver/work-x64-7.1/install/var/packages/synocli-videodriver/target/lib/libocloc.so
```
Notice that all libraries and binaries do go under the install prefix directory... except the `intel.icd`

th0ma7, 2024-09-30:
To add to this, `ocl-icd` has the following switches that I use:
```
CONFIGURE_ARGS  = --enable-custom-vendordir=$(INSTALL_PREFIX)/etc/OpenCL/vendors
CONFIGURE_ARGS += --enable-custom-layerdir=$(INSTALL_PREFIX)/etc/OpenCL/layers
```

JablonskiMateusz, 2024-10-02:
Output binaries are expected to follow `INSTALL_PREFIX` as that's the place where they are installed.
The icd file is not expected to follow `INSTALL_PREFIX`. In regular installation it is in `/etc/OpenCL/vendors/` independently on distro. That's a part of `cl_khr_icd` extension.

th0ma7, 2024-10-02:
Synology contraints packages to only be installed under `/var/packages/<package_name>`.
Therefore the `--enable-custom-vendordir=` and `--enable-custom-layerdir=` as promoted by `ocl-icd` allows me to change that at configuration time, leading to an install matching limitations of the platform I'm working with.

Having a similar option (or vendordir to follow install prefix option) for intel compute runtime would be welcomed.

And really this is the only thing left from this opened issue.  Otherwise help received so far allowed me to build and publish a functional SynoCommunity driver package and link ffmpeg to it to achieve opencl acceleration.  Feel free to close if there is no intention for implementing such option as I have my work-around that in my code at post install time.

JablonskiMateusz, 2024-10-02:
Sorry, I don't really understand what is expected here. OCL loader in default config is looking in `/etc/OpenCL/vendors/` and also in default case this can be overridden by env variable. 
`DESTDIR` point to root location of target installation and therefore we put icd file in `$DESTDIR/etc/OpenCL/vendors/`.
Are you expecting some cmake flag to allow putting icd file to `$DESTDIR/$NEO_CUSTOM_ICD_DIR`? If so, then you would also need to either compile ocl-icd with the custom vendor dir or set env to point custom vendor dir.

JablonskiMateusz, 2024-10-03:
@th0ma7  I'm looking at the code and I see there is a cmake variable that is handling that already in our code - `OCL_ICD_VENDORDIR` 
https://github.com/intel/compute-runtime/blob/master/package.cmake#L46
please try to use that

th0ma7, 2024-10-05:
indeed this works out as expected.  many thnx for the support, closing!

closed 2024-10-05 by th0ma7