intel/compute-runtime #759: Different compute-runtime version lead to different xpu memory usage

gc-fu, 2024-08-28:
Hi, I am running some xpu workload and found that different compute runtime will lead to different xpu memory usage.

When using version https://github.com/intel/compute-runtime/releases/tag/23.17.26241.22, the memory usage on Arc A770 is around 14.5G.

However, with latest compute runtime, the application will OOM on xpu memory, lead to xpu memory usage over 16286 MB.

My environment: 
driver version: 
```bash
ii  intel-i915-dkms                            1.24.2.17.240301.20+i29-1                         all          Out of tree i915 driver.
```

Driver related package version:

```bash
# These are installed in docker containers
# Driver related package version:
ii  intel-level-zero-gpu                           1.3.30049.6                             amd64        Intel(R) Graphics Compute Runtime for oneAPI Level Zero.
ii  level-zero-dev                                 1.14.0-744~22.04                        amd64        Intel(R) Graphics Compute Runtime for oneAPI Level Zero.
-----------------------------------------------------------------
```

gc-fu, 2024-08-28:
If needed, I can provide the steps for reproducing the problem.

JablonskiMateusz, 2024-08-28:
@gc-fu  could you please narrow-down which release changed the behavior? There are many releases in between. It would be great if you find which release caused the issue